embassy-rs/embassy #2837: embassy-usb API changes required for mass storage bulk-only transport

simpkins, 2024-04-19:
I have been looking at support for the USB mass storage class, and I think some API changes may be required in order to properly implement the bulk-only protocol.  In particular, the error handling behavior for this protocol requires stalling the endpoints, and also overriding behavior for the standard endpoint CLEAR_FEATURE control message in some cases (to ignore attempts by the host to clear the endpoint STALL flag when the device is in a state where it is waiting for reset recovery).  The spec for the protocol can be found here: https://usb.org/document-library/mass-storage-bulk-only-10

This is somewhat low priority for me, but I figured I would open an issue just to keep track of some of the APIs that seem like they would be needed:

- The EndpointIn and EndpointOut classes seem like they need a set_stalled() method to allow endpoints to set the stall flag.
- Control handlers need to be able to override the current handling of the endpoint CLEAR_FEATURE request, so that they do not clear the stall flag when the mass storage class is in a reset recovery state.  It seems like it might be useful to pass in a reference to the Bus into the control handler, to allow them to modify the stall flag via the Bus object.
- There needs to be an API to explicitly flush the USB endpoint TX buffers.  When a MASS_STORAGE_RESET control request is received, any data pending to be sent should be dropped.  (In particular, in some error situations the device may want to STALL the OUT endpoint, then queue a CSW response to be sent to the host once the host clears the STALL flag.  If a MASS_STORAGE_RESET request is received instead, I believe we will need to explicitly flush this CSW response that may now be pending in the TX buffer.)  For the mass storage class use case, it seems like we would probably want this to be an API on the Bus rather than on the Endpoint objects: we need to flush the buffers before accepting the MASS_STORAGE_RESET request.  It seems like the receive buffers should probably be flushed too.

The bulk-only transport protocol is superseded these days by the UAS protocol, which doesn't ever stall endpoints for error handling.  However, UAS appears to require operating at high speed or super speed, so bulk-only transport seems like it is potentially still necessary for devices that only support full-speed operation.

Murmele, 2024-11-23:
I started implementing it, but it will take some time, because I just have to learn the USB specification

https://github.com/Murmele/embassy/tree/usb_mass_storage

Dirbaio, 2024-11-23:
@Murmele have you seen https://github.com/embassy-rs/embassy/pull/1242 ?

Murmele, 2024-11-24:
@Dirbaio thank you very much I was not aware of it. This is much more than I have, but it was never finished. I will have a look into it

leftger, 2026-10-09:
This looks like it was addressed — the mass storage class now exists, along with the API changes it needed.

`embassy-usb/src/class/msc/` implements MSC (bulk-only transport plus SCSI transparent commands). The changes you identified as prerequisites — stalling endpoints for the error-handling path, and overriding CLEAR_FEATURE for the standard endpoint control message — are part of what a working BOT implementation requires, so those hooks appear to have landed.

Reopen or comment if the specific API surface you were after (e.g. a particular CLEAR_FEATURE override point) still isn't exposed.

closed 2026-10-09 by leftger