embassy-rs/embassy #2847: could not find `main` in `embassy_executor`

Gibbz, 2024-04-21:
Trying to get a simple example working with embassy.
On my esp32c3 I have the following main.rs
I want to get the sneak-peak section on the redme for embassy running.

```
error[E0433]: failed to resolve: could not find `main` in `embassy_executor`
  --> src/main.rs:12:21
   |
12 | #[embassy_executor::main]
   |                     ^^^^ could not find `main` in `embassy_executor`
```

main.rs
```
#![no_main]
#![feature(type_alias_impl_trait)]

use embassy_time::{Duration, Timer};
use embassy_executor::Spawner;

#[embassy_executor::main]
async fn main(spawner: Spawner) {
    // It is necessary to call this function once. Otherwise some patches to the runtime
    // implemented by esp-idf-sys might not link properly. See https://github.com/esp-rs/esp-idf-template/issues/71
    esp_idf_svc::sys::link_patches();

    // Bind the log crate to the ESP Logging facilities
    esp_idf_svc::log::EspLogger::initialize_default();

    spawner.spawn(run()).ok();
    log::info!("Hello, world!");
}

#[embassy_executor::task]
async fn run() {
    loop {
        log::info!("Hello world from embassy using esp-hal-async!");
        Timer::after(Duration::from_millis(1_000)).await;
    }
}
```

and cargo.toml
```
[package]
name = "stdtest"
version = "0.1.0"
authors = ["Bronson <bronsonmathews@gmail.com>"]
edition = "2021"
resolver = "2"
rust-version = "1.71"

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = ["std", "embassy", "esp-idf-svc/native"]

pio = ["esp-idf-svc/pio"]
std = ["alloc", "esp-idf-svc/binstart", "esp-idf-svc/std"]
alloc = ["esp-idf-svc/alloc"]
nightly = ["esp-idf-svc/nightly"]
experimental = ["esp-idf-svc/experimental"]
embassy = ["esp-idf-svc/embassy-sync", "esp-idf-svc/critical-section", "esp-idf-svc/embassy-time-driver"]

[dependencies]
log = { version = "0.4", default-features = false }
esp-idf-svc = { version = "0.48", default-features = false }
embassy-time = "0.3.0"
embassy-executor = "0.5.0"

[build-dependencies]
embuild = "0.31.3"

```

Dirbaio, 2024-04-21:
you have to enable features `executor-thread` and `arch-riscv` in `embassy-executor`.

Gibbz, 2024-04-22:
ok great thanks!

closed 2024-04-22 by Gibbz