embassy-rs/embassy #2852: embassy_executor::task inside an implementation issues

Gibbz, 2024-04-22:
I'm trying to spawn a loop/run task from inside a module implementation. I want to pass in a reference to self so that I can manipulate the data in the struct.

Im dealing with the 2 errors in the comments, and not sure how to proceed. When i remove the line '#[embassy_executor::task]' the first error is fixed, but guess I need it...

Full source code here: https://github.com/bit-shift-io/bike-aid/tree/main/bike-aid-mcu-rs

Code bellow:

```
use crate::trip::TripTrait;
use embassy_executor::{Executor, Spawner};
use embassy_time::{Duration, Timer};
use static_cell::StaticCell;

pub struct Clock {
    pub active: bool,
    pub start_time: u32,
    interval: u8,
    last_interval: u8,
    executor:  StaticCell<Executor>, // low priority task
    spawner: Spawner
}

impl Clock {
    pub fn new(spawner: Spawner) -> Self {
        Self {
            active: false,
            start_time: 0,
            interval: 0,
            last_interval: 0,
            executor: StaticCell::new(),
            spawner
        }
    }

    #[embassy_executor::task]
    async fn run(&mut self) {   //<-- ERROR: task functions must not have receiver arguments
        let mut count = 0;
        loop {
            log::info!("Clock Count: {}", count);
            count += 1;
            Timer::after(Duration::from_millis(60_000)).await;
        }
    }
}

impl TripTrait for Clock  {
    fn start(&mut self) {
        self.active = true;
        log::info!("starting clock");
        self.spawner.spawn(self.run()).ok(); //<-- ERROR: no method named `run` found for mutable reference `&mut trip::clock::Clock` in the current scope
method not found in `&mut Clock`
    }

    fn stop(&mut self) {
        self.active = false;
    }
}
```

Dirbaio, 2024-04-22:
you can't make methods be async tasks. you have to use a free-standing function:

```rust
#[embassy_executor::task]
async fn run(clock: Clock) { // pass the clock as an argument
    let mut count = 0;
    loop {
        log::info!("Clock Count: {}", count);
        count += 1;
        Timer::after(Duration::from_millis(60_000)).await;
    }
}
```

additionally, if you want the clock to share state between the run task and other tasks, you'll have to use `&'static Clock`  and use mutexes/atomics for the state.

Gibbz, 2024-04-22:
Ok thanks!

closed 2024-04-22 by Gibbz