embassy-rs/embassy #2867: Unable to create a microsecond accurate timer

ckellar33, 2024-04-25:
Using the blinky example, modified to run in an executor task, I cannot get the timer to be accurate to a millisecond. Best I have seen is around ~100us. I tried converting it to an interrupt executor, but still same results. 

Any thoughts? 

Is there an option to run a low-level timer?

Thank you,

lulf, 2024-04-25:
Which chip/hal are you using? You can configure the system tick rate setting a feature on the embassy-time-driver crate: https://docs.embassy.dev/embassy-time-driver/git/default/index.html#feature-flags

If you want to use a timer directly (i.e. you want a higher/lower freq timer compared to the system one), you can usually do that, but that depends on the HAL you are using. Example for stm32: https://docs.embassy.dev/embassy-stm32/git/stm32c011d6/timer/low_level/struct.Timer.html

ckellar33, 2024-11-15:
When running in release mode, the best I see is 5us plus whatever I delay in the task. If I wait 5us, the total duration of the led flash is 10us. Thoughts?

semiversus, 2025-03-05:
My first guess would be overhead. In a microsecond the processor can run 180 assembly commands (maximum) in your case. Having an async executer and maybe debug profile it's not a lot of work which can be done. Try if `--release` flag makes any difference.