embassy-rs/embassy #2868: Unable to run the wifi_esp_hosted demo on nRF52833_xxAA chip

BleemIs42, 2024-04-25:
```
cargo run --release --bin wifi_esp_hosted
    Finished release [optimized + debuginfo] target(s) in 0.56s
     Running `probe-rs run --chip nRF52833_xxAA target/thumbv7em-none-eabi/release/wifi_esp_hosted`
      Erasing ✔ [00:00:03] [##################################] 120.00 KiB/120.00 KiB @ 37.28 KiB/s (eta 0s )
  Programming ✔ [00:00:05] [##################################] 120.00 KiB/120.00 KiB @ 22.18 KiB/s (eta 0s )    Finished in 8.663s
 WARN defmt_decoder::log::format: logger format contains timestamp but no timestamp implementation was provided; consider removing the timestamp (`{t}` or `{T}`) from the logger format or provide a `defmt::timestamp!` implementation
0.000000 INFO  Hello World!
└─ wifi_esp_hosted::____embassy_main_task::{async_fn#0} @ src/bin/wifi_esp_hosted.rs:47
0.000854 DEBUG wait for init event...
└─ embassy_net_esp_hosted::control::{impl#0}::init::{async_fn#0} @ /Users/shuanghuiyan/Workspace/my-projects/embassy/embassy-net-esp-hosted/src/fmt.rs:130
0.000946 DEBUG resetting...
└─ embassy_net_esp_hosted::{impl#1}::run::{async_fn#0} @ /Users/shuanghuiyan/Workspace/my-projects/embassy/embassy-net-esp-hosted/src/fmt.rs:130
1.176239 TRACE rx: [00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]
└─ embassy_net_esp_hosted::{impl#1}::handle_rx @ /Users/shuanghuiyan/Workspace/my-projects/embassy/embassy-net-esp-hosted/src/fmt.rs:117
1.176849 TRACE rx: [00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]
└─ embassy_net_esp_hosted::{impl#1}::handle_rx @ /Users/shuanghuiyan/Workspace/my-projects/embassy/embassy-net-esp-hosted/src/fmt.rs:117
```

Dirbaio, 2024-04-25:
Have you connected an ESP32 module to the microbit, and flashed it following the instructions in the [esp-hosted](https://github.com/espressif/esp-hosted) repo?

BleemIs42, 2024-04-26:
Sorry, I didn't understand `connected an ESP32 module to the microbit`. I only have the micobiit board, it seems it can connect to wifi independ? 
My board hardware list: https://tech.microbit.org/hardware/

Dirbaio, 2024-04-26:
The microbit doesn't have wifi, only bluetooth. The `wifi_esp_hosted` example needs an external ESP32 module to add wifi capabilities, for example [this one](https://docs.espressif.com/projects/esp-idf/en/stable/esp32c3/hw-reference/esp32c3/user-guide-devkitm-1.html). It won't work with just the microbit.