embassy-rs/embassy #2890: some examples overflowing

pkoevesdi, 2024-04-29:
I saw it in the example https://github.com/embassy-rs/embassy/blob/679160a1c573709ccf2c54755e69ea9e1b5a209e/examples/stm32h7/src/bin/can.rs: the variable `i` is overflowing, the program is stopping / crashing (?) on the target.
I suggest to replace `i += 1;` with `i = i.wrapping_add(1);` to make it run forever, as probably expected.
Also, in several other examples:
https://github.com/search?q=repo%3Aembassy-rs%2Fembassy+%22i+%2B%3D+1%22+path%3A%2F%5Eexamples%5C%2F%2F&type=code

Dirbaio, 2024-04-29:
sounds a nice fix. can you send a PR?

YgorSouza, 2024-05-01:
The [clippy::arithmetic_side_effects](https://rust-lang.github.io/rust-clippy/master/index.html#/arithmetic_side_effects) lint catches that sort of thing. Maybe it could be enabled in some places.

Dirbaio, 2024-05-01:
fixed in #2891

closed 2024-05-01 by Dirbaio