embassy-rs/embassy #2896: Mismatch between online documentation and what cargo installs

vasilNnikolov, 2024-04-30:
I've tried to start a project for the RP2040, following the instructions [here](https://embassy.dev/book/dev/new_project.html), specifically using the [embassy-template](https://github.com/lulf/embassy-template) project. However,  there seems to be a mismatch between the version cargo installs with `cargo add embassy-rp`, which I think is from the tag [embassy-rp-v0.1.0](https://github.com/embassy-rs/embassy/tree/embassy-rp-v0.1.0), and the version whose documentation is in [the crate docs](https://docs.embassy.dev/embassy-rp/git/rp2040/index.html). I think the problem is that the docs website shows a later version of the library than the one tagged in this repo.

I found this as in both the current latest `main` commit of embassy (ecc910b76dbfa2064f42e6917a7b5654a89b81ed) and in the docs, the `embassy_rp::gpio::Output` struct takes only a lifetime, but in the version in the aforementioned tag, the struct takes both a lifetime and a generic type parameter.

#### in latest main, and in docs
```rust

pub struct Output<'d> {
    pin: Flex<'d>,
}
```
#### in tag embassy-rp-v0.1.0
```rust

pub struct Output<'d, T: Pin> {
    pin: Flex<'d, T>,
}
```
I know this can be avoided by specifying the latest commit in `Cargo.toml`, but it will be great is there is a way to view the documentation of the crates that are installed by default.

Dirbaio, 2024-04-30:
https://docs.embassy.dev hosts docs for the `git` version of the crate, as you can see in the top header.

to view docs for the crates.io release v0.1.0 use https://docs.rs/embassy-rp

vasilNnikolov, 2024-04-30:
Nice, thanks for the quick response!

closed 2024-04-30 by vasilNnikolov