embassy-rs/embassy #2906: stm32 i2cv1 async regression, silently reads back 0xFF

MabezDev, 2024-05-03:
Will need to do a bisect to find the offending commit, but essentially somewhere between the crates.io release and current main there is a regression where i2c doesn't fail it just reads back 0xFF. This was produced on an F2 board, but it more than likely applies to other chips with i2cv1.

Blocking of course works fine.

jamesmunns, 2024-05-12:
Might want to reference some of the PRs mentioned here: https://github.com/embassy-rs/embassy/issues/2372

SpencerSharkey, 2024-07-26:
i'm seeing the same behavior with i2cv2. blocking works, async doesn't- but only for pmbus devices in my experience so far

leftger, 2026-11-16:
Closing due to inactivity and no reproduction/bisect. Both `i2cv1` and `i2cv2` async paths have received several fixes since mid-2024 (including `v1` `write_read` BTF-before-RESTART in 1531266a1 and `v2` async read/transaction/DMA fixes in c50ffa852, e4e1e92c1, and da3009a7b). If you still observe async reads silently returning `0xFF` on current `main`, please open a new issue with a minimal reproducer (target chip, pins/DMA channels, and bus trace/code).

closed 2026-11-16 by leftger