embassy-rs/embassy #2913: Uart write work but not read fn

AnthonyYahtec, 2024-05-06:
The function **uart.read()** don't work in usart i use a STM32L431RC.

`
bind_interrupts!(struct Irqs {
    USART1 => usart::InterruptHandler<peripherals::USART1>;
});

#[embassy_executor::main]
async fn main(spawner: Spawner) {
    let device_config = embassy_stm32::Config::default();
    let p = embassy_stm32::init(device_config);
    let config = new_uart_config(9600u32, DataBits::DataBits8, Parity::ParityNone, StopBits::STOP1);
    let mut uart = Uart::new(p.USART1, p.PA10, p.PA9, Irqs, p.DMA1_CH4, p.DMA1_CH5, config).unwrap();

    info!("current usart1 priority: {:?}", Interrupt::USART1.get_priority()); //P0

    loop {
        let mut buf = [0u8; 128];
        // uart.blocking_write(&[0x02, 0x01]);

        let _n = uart.write(&mut [0x01, 0x02]).await;

        Timer::after_millis(200).await;
        uart.read_until_idle(&mut buf).await.unwrap();

        //Traitez les données reçues ici
        println!("Received data: {:?}", buf);

        Timer::after_millis(200).await;
    }}
`

lulf, 2024-05-06:
Keep in mind that unless you enable hardware flow-control, the uart read might miss the data, because the basic usart does not do any buffering and doesn't enable RX until the read_until_idle() is called. To increase the likelyhood that it will read the input, try configuring the buffered uart instead. In any case, using hardware flow control is the best way to guarantee not missing any data.

AnthonyYahtec, 2024-05-06:
This also don't work for read but work for write again
    let mut tx_buffer = [0u8; 4096];
    let mut rx_buffer = [0u8; 4096];
    let mut uart = BufferedUart::new(p.USART1, Irqs, p.PA10, p.PA9, &mut tx_buffer, &mut rx_buffer, config).unwrap();
    info!("current usart1 priority: {:?}", Interrupt::USART1.get_priority());
    // Interrupt::USART1.set_priority(interrupt::Priority::P0);
    //let mut uart = Uart::new(p.USART1, p.PA10, p.PA9, Irqs, NoDma, NoDma, config).unwrap();

    loop {
        let mut buf = [0u8; 128];
        // uart.blocking_write(&[0x02, 0x01]);

        //let _n = uart.write(&mut [0x01, 0x02]).await;
        uart.read(&mut buf).unwrap();

        //Timer::after_millis(200).await;
        // uart.read(&mut buf).await.unwrap();

        //Traitez les données reçues ici
        println!("Received data: {:?}", buf);

        Timer::after_millis(200).await;
    }
}

AnthonyYahtec, 2024-05-13:
This don't work anymore there is a feature to activate uart write ?