embassy-rs/embassy #2926: Problem with `embassy_stm32::ospi::Ospi::new_*`

pkoevesdi, 2024-05-10:
I have problems creating a new OSPI driver for a quadspi external chip. I get 
`the trait 'embassy_stm32::ospi::*<embassy_stm32::peripherals::OCTOSPI1>' is not implemented for '*'` for all 6 Pins. Am I doing something wrong here or is it a bug?
I try it with the current git version of embassy.

If I look at https://github.com/embassy-rs/embassy/blob/main/examples/stm32f7/src/bin/qspi.rs, I think I have cooked it down and abstracted from the mcu- and flash-specific parts.
Also, I see the function https://github.com/embassy-rs/embassy/blob/4b4777e6bb813dbde3f8ec05ff81cadbcb41bee0/embassy-stm32/src/qspi/mod.rs#L88
equivalent to
https://github.com/embassy-rs/embassy/blob/4b4777e6bb813dbde3f8ec05ff81cadbcb41bee0/embassy-stm32/src/ospi/mod.rs#L259
so I assume, I can use the qspi example as a pattern for ospi too?

That's the minimal code which I'd expect to build:
```
#![no_std]
#![no_main]

use embassy_executor::Spawner;
use embassy_stm32::dma::NoDma;
use embassy_stm32::ospi::{Config, Ospi};
use panic_probe as _;

#[embassy_executor::main]
async fn main(_spawner: Spawner) {

    let p = embassy_stm32::init(Default::default());
    let mut config = Config::default();

   let qspi = Ospi::new_quadspi(
        p.OCTOSPI1, p.PA3, p.PC9, p.PC10, p.PE2, p.PA6, p.PC11, NoDma, config,
    );

}
```

pkoevesdi, 2024-05-10:
I checked https://github.com/embassy-rs/stm32-data-generated/tree/main and found all pins seemingly correctly assigned, for instance:
https://github.com/embassy-rs/stm32-data-generated/blob/main/data/chips/STM32H730VB.json#L5115

Mirror0, 2024-05-10:
I brefly check the `stm32-data-generated` file you provided and compare it to `QSPI` declarations with `STM32F7` board that I used.

I see some differences:
- Names in `pin_trait_impl` section of `build.rs` are different in  provided H7 file for `OSPI` pins then compared to corresponding `QSPI` pins for F7 board.
https://github.com/embassy-rs/embassy/blob/9cbbedef793d619c659c6a81080675282690a8af/embassy-stm32/build.rs#L1021 
      **Example:** 
      F7 board has `BK1_I03` pin as `PA1` under `qspi` section and there is `BK1_I03` under `qspi` in `build.rs`, but H7 board has `P1_CLK` pin as `PA3` under `octospim` but in `ospi` there's only `CLK`.
- Additionally in H7 file pins that you pointed are under `octospim` registry kind in opposition to F7 file where QSPI pins are under `qspi` register kind which corresponds with naming in `build.rs`. This also may be a problem as above the `pin_trait_impl` section there is this comment ` (kind, signal) => trait`, I guess the `kind` is from `registers` section and `signal` is from `pins` section

This can be the cause for not implemented error, but I'm not knowledgeable about embassy build process and embedded programming to answer this with certainty.

pkoevesdi, 2024-05-10:
Thanks for the hints, I'll follow them. But actually, this looks like something that @Dirbaio has to look at, doesn't it?

Mirror0, 2024-05-10:
I think it would be best if he takes a look at this, as he is more experienced in this field :)

pkoevesdi, 2024-05-10:
So, it seems to be about the OCTOSPI manager that sits between the physical pins i.e.:
```
OCTOSPIM_P1_ CLK
OCTOSPIM_P1_IO0
OCTOSPIM_P1_IO1
OCTOSPIM_P1_IO2
OCTOSPIM_P1_IO3
OCTOSPIM_P1_NCS
```
and the internal OCTOSPI1 virtual port.
In reset state, the both are mapped 1:1.

From the [reference manual](https://www.st.com/resource/en/reference_manual/rm0468-stm32h723733-stm32h725735-and-stm32h730-value-line-advanced-armbased-32bit-mcus-stmicroelectronics.pdf), p. 964:
[image]

pkoevesdi, 2024-05-11:
Here I accidently found a micrppython guy addressing the same issue, seemingly:
https://github.com/micropython/micropython/issues/12517
May that helps for embassy too?