embassy-rs/embassy #2936: Define a regular crates-io release cycle

jamesmunns, 2024-05-13:
It would be good to commit to a more regular crates-io release cycle, in order to get more people away from patches when possible.

It might be good to set up some regular cadence, like every week or two, if there have been any changes.

I don't think we necessarily need to commit to a more stringent stability guarantee, it could be a (breaking) point release every time for the foreseeable future.

jamesmunns, 2024-05-20:
Chatted with @Dirbaio, there's loose interest, before we can do that, we need to make the following docs/process things happen:

- [ ] Get an inventory of all crates in the embassy workspace, including intra-crate deps (so we know what order to release in)
- [ ] Add a release team to embassy on GH, make sure all crates-io crates have permissions for that team
- [ ] Add people to that team (who?)
- [ ] Write a checklist for periodic releases, like reving version numbers, updating changelogs, tagging commits, etc.
- [ ] Write in the docs somewhere what the planned cadence is (1/2/3/4/5/6 weeks)
- [ ] Do a first release with "all hands on deck", update with any missed details

jamesmunns, 2024-05-20:
## Crate Inventory

```sh
$ git log | head -n 1                             
commit b33ef0d7dd4eb9124626ca5242cab57906b3007c
$ find . | rg 'Cargo.toml$' | rg -v '/examples/' | rg -v '/tests/'
./embassy-net-ppp/Cargo.toml
./embassy-usb-driver/Cargo.toml
./embassy-stm32/Cargo.toml
./embassy-net-esp-hosted/Cargo.toml
./cyw43-pio/Cargo.toml
./embassy-time/Cargo.toml
./embassy-executor/Cargo.toml
./embassy-time-queue-driver/Cargo.toml
./embassy-boot-nrf/Cargo.toml
./embassy-net-driver-channel/Cargo.toml
./embassy-rp/Cargo.toml
./embassy-hal-internal/Cargo.toml
./embassy-sync/Cargo.toml
./embassy-usb/Cargo.toml
./embassy-nrf/Cargo.toml
./embassy-futures/Cargo.toml
./embassy-net-wiznet/Cargo.toml
./embassy-usb-logger/Cargo.toml
./embassy-usb-synopsys-otg/Cargo.toml
./embassy-net-driver/Cargo.toml
./embassy-net-tuntap/Cargo.toml
./embassy-usb-dfu/Cargo.toml
./embassy-boot-rp/Cargo.toml
./embassy-net-adin1110/Cargo.toml
./embassy-boot/Cargo.toml
./embassy-net-enc28j60/Cargo.toml
./embassy-time-driver/Cargo.toml
./embassy-embedded-hal/Cargo.toml
./embassy-boot-stm32/Cargo.toml
./cyw43/Cargo.toml
./embassy-net/Cargo.toml
./embassy-executor-macros/Cargo.toml
./embassy-stm32-wpan/Cargo.toml
```

Still todo, figuring out the inverted dep tree order.

YgorSouza, 2024-09-13:
[egui](https://github.com/emilk/egui) might have some good ideas to borrow from. It doesn't have a rigid release train like Rust, but there is a new major release for all crates consistently every 2-3 months, with a few patches in between for hotfixes. There is some amount of automation to speed things up. If there are no changes, there are no changes (see the [ecolor changelog](https://github.com/emilk/egui/blob/master/crates/ecolor/CHANGELOG.md) for example). All crates just stay on the same version regardless, so it is easier to reason about and to automate.

The process is described in <https://github.com/emilk/egui/blob/master/RELEASES.md>