embassy-rs/embassy #2942: STM32 GPIOv1 input pins being set incorrectly (for SPI)

tobywf, 2024-05-15:
As far as I can tell, the GPIO MODE bits end up being set incorrectly for the MISO pin (input) when SPI is being set up for devices that use `gpio_v1`. It could also affect other functions that declare "alternate function" input pins.

This behaviour is also present in `embassy_stm32` version `0.1.0` on crates.io. The code is slightly different since the `new_pin!()` macro doesn't exist, but the calls end up being the same.

I don't really understands why this doesn't usually manifest in a bug; I can only assume that for most genuine STM chips if the CNF bits aren't changed, then changing the MODE bits doesn't change the pin's configuration mode.

Please bear with me, this has been hard to trace due to all the indirections. 

---

TL;DR: The issue is that pins are being configured "twice", using `set_as_af_pull()` and `set_speed()`. For input pins, the latter is incorrect.

On page 161 the reference manual [RM0008](https://www.st.com/resource/en/reference_manual/rm0008-stm32f101xx-stm32f102xx-stm32f103xx-stm32f105xx-and-stm32f107xx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) Rev21 for STM32F101xx, STM32F102xx, STM32F103xx, STM32F105xx and STM32F107xx, table 20. "Port bit configuration table" indicates that for input pins, the MODE bits ought to be `00`.

[image]

STM32F10X devices all seem to use use `gpio_v1`, e.g. `STM32F103RE` in [`stm32-metapac/src/chips/stm32f103re/pac.rs`](https://github.com/embassy-rs/stm32-data-generated/blob/bedafad200aaaf1c879505d8f6253911f767c9b2/stm32-metapac/src/chips/stm32f103re/pac.rs#L86)

```rust
#[path="../../peripherals/gpio_v1.rs"] pub mod gpio;
```

All SPI pins are set up via the `new_pin!()` macro for both blocking and async, but MISO is the interesting one as it is an input:

https://github.com/embassy-rs/embassy/blob/ea70b440cd1035f28c3f332a2f72d7fa42ac995d/embassy-stm32/src/spi/mod.rs#L412

https://github.com/embassy-rs/embassy/blob/ea70b440cd1035f28c3f332a2f72d7fa42ac995d/embassy-stm32/src/spi/mod.rs#L490

Initially, in the `new_pin!()` macro, the MODE bits are set correctly via the `pin.set_as_af_pull()` method:

https://github.com/embassy-rs/embassy/blob/ea70b440cd1035f28c3f332a2f72d7fa42ac995d/embassy-stm32/src/macros.rs#L59-L72

Expanded, this is:

```rust
miso.set_as_af_pull(miso.af_num(), AFType::Input, Pull::None);
miso.set_speed(Speed::VeryHigh);
```

Looking at `set_as_af_pull()` for `gpio_v1`, it takes the `AFType::Input` branch and calls `set_mode()`:

https://github.com/embassy-rs/embassy/blob/ea70b440cd1035f28c3f332a2f72d7fa42ac995d/embassy-stm32/src/gpio.rs#L579-L620

```rust
w.set_mode(n % 8, vals::Mode::INPUT); // n == pin
```

Per the [`stm32-metapac/src/peripherals/gpio_v1.rs`](https://github.com/embassy-rs/stm32-data-generated/blob/bedafad200aaaf1c879505d8f6253911f767c9b2/stm32-metapac/src/peripherals/gpio_v1.rs#L115-L119), this is the following (edited for readability):

```rust
pub enum Mode {
    /// Input mode (reset state)
    INPUT = 0b00, // 0x0
    /// Output mode 10 MHz
    OUTPUT10MHZ = 0b01, // 0x01
    /// Output mode 2 MHz
    OUTPUT2MHZ = 0b10, // 0x02
    /// Output mode 50 MHz
    OUTPUT50MHZ = 0b11, // 0x03
}
```

So far, so good. However, `set_speed()` is called immediately after `set_as_af_pull()`. This is where it goes wrong. `Speed` maps to `vals::Mode` for `gpio_v1`:

https://github.com/embassy-rs/embassy/blob/ea70b440cd1035f28c3f332a2f72d7fa42ac995d/embassy-stm32/src/gpio.rs#L275-L286

This eventually results in the MODE bits being set to something other than `00`, so not an input any more.

https://github.com/embassy-rs/embassy/blob/ea70b440cd1035f28c3f332a2f72d7fa42ac995d/embassy-stm32/src/gpio.rs#L665-L678

TL;DR: For inputs, `set_speed()` overwrites the MODE bits correctly set by `set_as_af_pull()`.

Seems like an obvious solution is to fuse `set_speed()` into `set_as_af_pull()`, i.e. `set_as_af_pull` should also take the speed since the CNF and MODE bits aren't independent (for `gpio_v1` at least).

For `gpio_v1`, this would result in the correct mode for `AFType::Input` as the speed would be ignored. And for `AFType::OutputPushPull` and `AFType::OutputOpenDrain`, the speed could be set immediately instead of being hardcoded to `vals::Mode::OUTPUT50MHZ`:

https://github.com/embassy-rs/embassy/blob/ea70b440cd1035f28c3f332a2f72d7fa42ac995d/embassy-stm32/src/gpio.rs#L607-L618

(For `gpio_v2`, the current `set_speed()` functionality to modify the `ospeedr` register would have to be done in `set_as_af_pull()` also.)

joelsa, 2024-05-22:
Hi, thank you very much for reporting this and tracking it down so nicely! This is very easy to understand.

I need this for https://github.com/embassy-rs/embassy/issues/2798, since in order to determine the current pin configuration, the MODE bits need to be set correctly.

My initial idea was a very minimal and non-invasive solution: Adding a `Speed::Input` (or maybe `Speed::None`) to the Speed enum so that no wrong value will be passed to begin with. Passing any type of speed for an input is misleading and should not be done in the first place. I still like this idea, but also dislike it because it makes it easy for users to make erros while copy & pasting, since it does not check for validity.

Better ideas imo:

- Modify the `new_pin` macro to not call `set_mode()` for inputs
- Have a `new_input` and `new_output` macro, the `new_input` macro does not take a Speed
- Move the `set_speed()` into `set_as_af_pull()` and check for input there
- Additionally: Move the `set_speed()` from `SealedPin` to `Output`, so that it is not implemented for Inputs. For gpio v2, speed is irrelevant for inputs. For gpio v1, the Mode bits should not be written to for an input that should stay an input.

I think a combination of the last three points is the cleanest and best solution and I will hand in a PR with that solution in the coming days.

tobywf, 2024-05-23:
Suggestion 3 ("Move the `set_speed()` into `set_as_af_pull()` and check for input there") is what I was proposing also.  That would be a breaking change though, not sure what the maintainers think.

Suggestion 1 (modifying the `new_pin!` macro) is something I also thought about. I didn't propose it since a) it doesn't get rid of setting the speed twice and b) I wasn't sure how `set_speed()` for `gpio_v2` behaves. I had a brief look at RM0090 for STM32F4, and it makes sense that a register called `ospeedr`/output speed register doesn't matter for inputs. And thinking about it, this might be a good solution because it doesn't require changing `set_as_af_pull()`.

Introducing the specific input/output macros would be cleaner, but maybe something like this is sufficient for now?

```rust
macro_rules! new_pin {
    ($name:ident, $aftype:expr) => {{
        new_pin!($name, $aftype, crate::gpio::Speed::Medium, crate::gpio::Pull::None)
    }};
    ($name:ident, $aftype:expr, $speed:expr) => {
        new_pin!($name, $aftype, $speed, crate::gpio::Pull::None)
    };
    ($name:ident, $aftype:expr, $speed:expr, $pull:expr) => {{
        let pin = $name.into_ref();
        pin.set_as_af_pull(pin.af_num(), $aftype, $pull);
        if $aftype != crate::gpio::low_level::AFType::Input {
            pin.set_speed($speed);
        }
        Some(pin.map_into())
    }};
}
```

Since the branch/comparison is between two values known at compile time, it ought to be optimised away.

As for moving `set_speed()` to `Output`, not sure if that's possible right now. Mainly since AF pins aren't currently `Output`...

closed 2024-05-24 by Dirbaio