embassy-rs/embassy #2944: Using embassy-boot on esp architectures

Sycrosity, 2024-05-15:
As there is to my knowledge not an available DFU bootloader on esp for no_std applications, I've been attempting to implement `embassy-boot` for ESP architectures, attempting to copy the layout of the `embassy-boot-*` crates - however, the way that the `embassy-boot-*` crates `Bootloader::load()` functions work (which load the application thats in the "active" partition from what I can understand) uses cortex_m functions which are (at least from my searching) not available in the xtensa and riscv crates. How could I go about implementing this (or finding out how to implement it) for esp platforms?

lulf, 2024-05-15:
I don't know the esp hal, but some searching through esp-hal I think I found some reset functions here https://github.com/esp-rs/esp-hal/blob/9edd098da56be11dde2ba7cd8ecffaaea23cb030/esp-hal/src/reset.rs#L106-L124

So the idea would be to replace the cortex-m bits with something like the above. Maybe @MabezDev can provide confirmation or additional info about reset.

Sycrosity, 2024-05-16:
Thank you for those! With some more tinkering, I have replaced the WatchdogFlash section of the bootloader, with suitable esp replacements (esp-storage is very useful for this) - however I still am not sure on the rest of the loading step (as before using the reset commands you have to modify the stack pointer). `xtensa_lx::set_stack_pointer` and `xtensa_lx::set_vecbase` seem like they could be helpful? (https://docs.rs/xtensa-lx/0.9.0/xtensa_lx/)

Sycrosity, 2024-10-15:
@benedikt-bartscher I sadly didn't get it to work, but if you do I'd love to see how you do it!