embassy-rs/embassy #2959: dyn_sender() and dyn_receiver() not found

vortex314, 2024-05-18:
````rust
static TXD_MSG: Channel<CriticalSectionRawMutex, ProxyMessage, 5> = Channel::new();
static RXD_MSG: Channel<CriticalSectionRawMutex, ProxyMessage, 5> = Channel::new();

  let clientSession = ClientSession::new(TXD_MSG.dyn_sender(), RXD_MSG.dyn_receiver());
    let uartActor = UartActor::new(uart0,TXD_MSG.dyn_receiver(),RXD_MSG.dyn_sender());

````
gives compiler errors , which I cannot get my head around why it fails.

```
error[E0599]: no method named `dyn_sender` found for struct `embassy_sync::channel::Channel` in the current scope
  --> src/main.rs:74:52
   |
74 |     let clientSession = ClientSession::new(TXD_MSG.dyn_sender(), RXD_MSG.dyn_receiver());
   |                                                    ^^^^^^^^^^ help: there is a method with a similar name: `sender`

error[E0599]: no method named `dyn_receiver` found for struct `embassy_sync::channel::Channel` in the current scope
  --> src/main.rs:74:74
   |
74 |     let clientSession = ClientSession::new(TXD_MSG.dyn_sender(), RXD_MSG.dyn_receiver());
   |                                                                          ^^^^^^^^^^^^ help: there is a method with a similar name: `receiver`

error[E0599]: no method named `dyn_receiver` found for struct `embassy_sync::channel::Channel` in the current scope
  --> src/main.rs:75:50
   |
75 |     let uartActor = UartActor::new(uart0,TXD_MSG.dyn_receiver(),RXD_MSG.dyn_sender());
   |                                                  ^^^^^^^^^^^^ help: there is a method with a similar name: `receiver`

error[E0599]: no method named `dyn_sender` found for struct `embassy_sync::channel::Channel` in the current scope
  --> src/main.rs:75:73
   |
75 |     let uartActor = UartActor::new(uart0,TXD_MSG.dyn_receiver(),RXD_MSG.dyn_sender());
   |                                                                         ^^^^^^^^^^ help: there is a method with a similar name: `sender`
````

Dirbaio, 2024-05-18:
these methods are available in `embassy-sync` from git. they're not in `embassy-sync` v0.5 from crates.io, they will be in the next release..

vortex314, 2024-05-18:
That's confusing as the embassy docs is documenting this. https://docs.embassy.dev/embassy-sync/git/default/channel/struct.Channel.html and docs.rs has broken links : https://docs.rs/crate/embassy-sync/0.5.0

Thanks for the answer @Dirbaio , I can probably figure it out further.

closed 2024-05-18 by vortex314