embassy-rs/embassy #2995: Stack Overrun on STM32F747ZG Nucleo Board with blinky example

droofe, 2024-05-24:
Hey there.

I've changed the Cargo.toml and config.toml to reflect the correct chip.

When I `cargo run --release --bin blinky` in `embassy/examples/stm32f7`, it appears that DefaultPreInit returns a stack end address that exceeds the memory map, causing the chip to hard fault after passing 0x2005ffe0.

```
Status: Halted(Unknown)
Core halted at address 0x0800513e
XPSR: 0x61000003
Currently handling exception 3
Hard Fault!
Return address (LR): 0xfffffff9
-> configurable priority exception has been escalated to hard fault!
	Bus Fault       - BFSR: 0x04
```

Dirbaio, 2024-05-24:
this is probaby due to https://github.com/embassy-rs/stm32-data/issues/301.

try specifying memory.x manually

droofe, 2024-05-24:
Ah interesting. Specifying a custom `memory.x` worked.

Dirbaio, 2024-05-24:
confirmed then. closing in favor of https://github.com/embassy-rs/stm32-data/issues/301.

closed 2024-05-24 by Dirbaio