embassy-rs/embassy #2997: USB HID Keyboard not working in BIOS (before Linux start)

dariogoetz, 2024-05-24:
MCU: STM32F401CC

I have written an embassy-based keyboard firmware for a keyboard. The firmware works well in Linux. However, the keyboard does not produce anything when the laptop asks for a system password when booting (before Linux starts).

I could reproduce the issue with the code from the examples directory `examples/stm32f4/src/bin/usb_hid_keyboard.rs` (adjusted for clocks of the STM32F401CC, otherwise vanilla): Within my Linux system, it produces an `a` when pulling PC13 up. In the password prompt before starting Linux, however, nothing happens.

I have another firmware based on `rtic`, `stm32f4xx-hal`, and `usb-device` that does not have this issue, so I would exclude it being a hardware issue.

Any ideas, what the issue could be?

dariogoetz, 2024-07-08:
Thank you very much for the feedback. I tried various different combinations, also setting different values for `config.device_class`  (the default value and `0x03`) and `config.composite_with_iads` while fixing `config.device_sub_class=0x01`. None worked, unfortunately.

When you wrote "f your device supports the boot protocol", what is the "device" you speak of? I am using an STM32F401CCU6 black pill board. Is that what you mean? Apart from that, the various requirements described in the F and G sections of the document that you linked refer to software-related things that are (or should be) implemented in `embassy-usb`. Maybe I got that wrong, though.

leftger, 2026-11-16:
Closing as resolved by #4691, which added HID Boot Protocol support (`HidSubclass::Boot`, `HidBootProtocol`, and `RequestHandler::{get,set}_protocol`) to `embassy-usb` and updated `examples/stm32f4/src/bin/usb_hid_keyboard.rs` to advertise Boot Keyboard subclass/protocol for BIOS/pre-boot compatibility.

closed 2026-11-16 by leftger