embassy-rs/embassy #2999: Unable to build CAN example for STM32F103

thebeekeeper, 2024-05-24:
Just getting started exploring embassy and embedded Rust in general, so sorry if this is a naive question.  I've designed a custom PCB with the STM32F103T4U on it.  I'm able to successfully program the device using the ST IDE, so I know it's wired up correctly.  I can also build and run the `blinky` example in the embassy repository. 

I need this board to communicate over CAN, but when I try to build the example under `examples/stm32f1/src/bin` I see the following error:

```
  = note: rust-lld: error:
          ERROR(cortex-m-rt): The .text section must be placed inside the FLASH memory.
          Set _stext to an address smaller than 'ORIGIN(FLASH) + LENGTH(FLASH)'

          rust-lld: error:
          ERROR(cortex-m-rt): The .text section must be placed inside the FLASH memory.
          Set _stext to an address smaller than 'ORIGIN(FLASH) + LENGTH(FLASH)'

          rust-lld: error:
          ERROR(cortex-m-rt): The .text section must be placed inside the FLASH memory.
          Set _stext to an address smaller than 'ORIGIN(FLASH) + LENGTH(FLASH)'

error: could not compile `embassy-stm32f1-examples` (bin "can") due to 1 previous error
```

So, my guess here is that the CAN example is too large to fit in my microcontroller.  I did try removing the `memory-x` feature and using a manually generated `memory.x` file and got the same error message.  I did try increasing the `FLASH` section length to 64K, and the example builds, but of course I can't flash it.

Does my guess sound correct that there is not enough memory for this example program?

Dirbaio, 2024-05-24:
Are you building with --release?

Dirbaio, 2024-05-24:
ah I missed you've edited the example for stm32f103t4 which only has 16kb of flash. That's quite small, you're going to have trouble getting things to fit.

Make sure to check the steps [here](https://embassy.dev/book/#_why_is_my_binary_so_big). I was able to get it to 10kb with lto, opt-level=z, panic_immediate_abort. Removing defmt gets it down to 8kb.

thebeekeeper, 2024-05-24:
Ok, that seems to work to get the hex file flashed to the device when I remove defmt and turn on more optimization.  Probably going to redesign around another micro, but its good to know the embedded Rust toolchain is working.

Didn't know about cargo size, very useful tool.

Thanks for your help!

closed 2024-05-24 by thebeekeeper