embassy-rs/embassy #3022: Release gpio Pins back to Peripharals

lgjonathan, 2024-05-30:
Hi, I have a usecase in stm32 that needs to use a pin both as a Output and  Uart. Is there a way to release the Peripharals from the `Uart` and `Output`?

Is using reborrow the only way to achieve this?

Dirbaio, 2024-05-30:
the way you do this is create the `Output` and the `Uart` with `&mut p.PA9`. When you drop the `Output` the borrow is released, so you can reuse the same pin with Uart.

If you create the Output or Uart with owned pins there's no way to get them back. The reason is the exact pin type is lost because the generic parameters for Output/Uart don't contain the pin numbers (and it's intentional that they don't, having too many generics is annoying). If we implemented this, they'd have to return the pins as `AnyPin`, which wouldn't be useful for e.g. creating an uart with it because that has to validate the pin is valid for UART.

If using `&mut p.PA9` is annoying due to how you've structured your app, you can also use `unsafe { PA9::steal() }` to get unhecked copies of the peripherals. It's then your responsibility to not use the same pin multiple times at the same time, but if you do that properly then this is a solid viable option.

lgjonathan, 2024-05-31:
Thanks, I'll look into it

closed 2024-05-31 by lgjonathan