embassy-rs/embassy #3026: When enabling DNS feature for rp pico W network stack never finish setup

ODCA117, 2024-05-31:
I'm trying to make a small program with DNS feature enabled on Rasberry pi Pico W but it never seem to come past the setup.
In the following program the led is never turned on. I have verified that it is while waiting for the stack to be up that the program gets stuck. I have tried to use a static configuration instead of DHCP with no success. Disabling the DNS feature for embassy-net seem to fix it. The problem is that I would like to be able to make a DNS Query.

main.rs:
```
#[embassy_executor::task(pool_size = 2)]
async fn wifi_task(runner: cyw43::Runner<'static, Output<'static>, PioSpi<'static, PIO0, 0, DMA_CH0>>) -> ! {
    runner.run().await
}

#[embassy_executor::task(pool_size = 2)]
async fn net_task(stack: &'static Stack<cyw43::NetDriver<'static>>) -> ! {
    stack.run().await
}

#[embassy_executor::main]
async fn main(spawner: Spawner) {

    let p = embassy_rp::init(Default::default());

    let fw = include_bytes!("../cyw43-firmware/43439A0.bin");
    let clm = include_bytes!("../cyw43-firmware/43439A0_clm.bin");

    // To make flashing faster for development, you may want to flash the firmwares independently
    // at hardcoded addresses, instead of baking them into the program with `include_bytes!`:
    //     probe-rs download 43439A0.bin --format bin --chip RP2040 --base-address 0x10100000
    //     probe-rs download 43439A0_clm.bin --format bin --chip RP2040 --base-address 0x10140000
    //let fw = unsafe { core::slice::from_raw_parts(0x10100000 as *const u8, 230321) };
    //let clm = unsafe { core::slice::from_raw_parts(0x10140000 as *const u8, 4752) };

    let pwr = Output::new(p.PIN_23, Level::Low);
    let cs = Output::new(p.PIN_25, Level::High);
    let mut pio = Pio::new(p.PIO0, Irqs);
    let spi = PioSpi::new(&mut pio.common, pio.sm0, pio.irq0, cs, p.PIN_24, p.PIN_29, p.DMA_CH0);

    static STATE: StaticCell<cyw43::State> = StaticCell::new();
    let state = STATE.init(cyw43::State::new());
    let (net_device, mut control, runner) = cyw43::new(state, pwr, spi, fw).await;
    spawner.spawn(wifi_task(runner)).unwrap();

    control.init(clm).await;
    control
        .set_power_management(cyw43::PowerManagementMode::PowerSave)
        .await;

    let config = Config::dhcpv4(Default::default());
    // Generate random seed
    let seed = 0x0123_4567_89ab_cdef; // chosen by fair dice roll. guarenteed to be random.

    // Init network stack
    static STACK: StaticCell<Stack<cyw43::NetDriver<'static>>> = StaticCell::new();
    static RESOURCES: StaticCell<StackResources<1>> = StaticCell::new();
    let stack = &*STACK.init(Stack::new(
        net_device,
        config,
        RESOURCES.init(StackResources::<1>::new()),
        seed,
    ));

    spawner.spawn(net_task(stack)).unwrap();

    match control.join_wpa2(WIFI_NETWORK, WIFI_PASSWORD).await {
        Ok(_) => (),
        Err(_) => {
            loop {
                control.gpio_set(0, true).await;
                Timer::after_millis(100).await;
                control.gpio_set(0, false).await;
                Timer::after_millis(100).await;
            }
        }
    }

    // Wait for DHCP, not necessary when using static IP
    while !stack.is_config_up() {
        Timer::after_millis(100).await;
    }

    // THIS IS NEVER REACHED
    control.gpio_set(0, true).await;

    // And now we can use it!
}
```

Cargo.toml:
```
[package]
edition = "2021"
name = "embassy-rp-examples"
version = "0.1.0"
license = "MIT OR Apache-2.0"

[dependencies]
embassy-embedded-hal = { version = "0.1.0", git = "https://github.com/embassy-rs/embassy", features = ["defmt"] }
embassy-sync = { version = "0.5.0", git = "https://github.com/embassy-rs/embassy",  features = ["defmt"] }
embassy-executor = { version = "0.5.0", git = "https://github.com/embassy-rs/embassy", features = ["task-arena-size-32768", "arch-cortex-m", "executor-thread", "executor-interrupt", "defmt", "integrated-timers"] }
embassy-time = { version = "0.3.0", git = "https://github.com/embassy-rs/embassy", features = ["defmt", "defmt-timestamp-uptime"] }
embassy-rp = { version = "0.1.0", git = "https://github.com/embassy-rs/embassy", features = ["defmt", "unstable-pac", "time-driver", "critical-section-impl"] }
embassy-usb = { version = "0.2.0", git = "https://github.com/embassy-rs/embassy", features = ["defmt"] }
embassy-net = { version = "0.4.0", git = "https://github.com/embassy-rs/embassy", features = [ "defmt", "medium-ethernet", "medium-ip", "tcp", "udp", "dns", "dhcpv4", "proto-ipv4", "proto-ipv6"] }
embassy-net-wiznet = { version = "0.1.0", git = "https://github.com/embassy-rs/embassy", features = ["defmt"] }
embassy-futures = { version = "0.1.0", git = "https://github.com/embassy-rs/embassy"}
embassy-usb-logger = { version = "0.1.0", git = "https://github.com/embassy-rs/embassy"}
cyw43 = { version = "0.1.0", path = "./cyw43", features = ["defmt", "firmware-logs"] }
cyw43-pio = { version = "0.1.0", path = "./cyw43-pio", features = ["defmt", "overclock"] }

defmt = "0.3"
defmt-rtt = "0.4"
fixed = "1.23.1"
fixed-macro = "1.2"

#cortex-m = { version = "0.7.6", features = ["critical-section-single-core"] }
cortex-m = { version = "0.7.7", features = ["inline-asm"] }
cortex-m-rt = "0.7.0"
panic-probe = { version = "0.3", features = ["print-defmt"] }
display-interface-spi = "0.4.1"
embedded-graphics = "0.7.1"
st7789 = "0.6.1"
display-interface = "0.4.1"
byte-slice-cast = { version = "1.2.0", default-features = false }
smart-leds = "0.3.0"
heapless = "0.8"
usbd-hid = "0.7.0"

embedded-hal-1 = { package = "embedded-hal", version = "1.0" }
embedded-hal-async = "1.0"
embedded-hal-bus = { version = "0.1", features = ["async"] }
embedded-io-async = { version = "0.6.1", features = ["defmt-03"] }
embedded-storage = { version = "0.3" }
static_cell = "2"
portable-atomic = { version = "1.5", features = ["critical-section"] }
log = "0.4"
pio-proc = "0.2"
pio = "0.2.1"
rand = { version = "0.8.5", default-features = false }

[profile.release]
debug = 2
```

Dirbaio, 2024-05-31:
Do you have a debug probe? this is probably a panic, and you can only see panic messages with a debug probe.

ODCA117, 2024-05-31:
I do not have any debug probe sadly,

ODCA117, 2024-06-19:
There where 2 issues, one was that I did not add enought network stack resources (StackResources<4> worked) and the other was that there where to many DNS servers so I had to increase that config (SMOLTCP_DNS_MAX_SERVER_COUNT = "5")

closed 2024-06-19 by ODCA117