embassy-rs/embassy #3030: [Feature] STM32 HAL required for `EXTI13` to implement `Into<embassy_stm32::exti::AnyChannel>`

FourLeafTec, 2024-06-01:
I have two pins that need to perform similar tasks in the interrupt, so I wrote the following code:

```
#[embassy_executor::task(pool_size = 4)]
pub async fn north_task(pin: AnyPin, ch: AnyChannel) {
    // Remove the lifetime constraint from the function input types
    let mut input = ExtiInput::new(pin, ch, Pull::Down);
    loop {
        input.wait_for_rising_edge().await;
        info!("Pressed!");
        input.wait_for_falling_edge().await;
        info!("Released!");
    }
}
#[embassy_executor::main]
async fn main(spawner: Spawner) {
    let p = embassy_stm32::init(Default::default());
    info!("Hello World!");
   
    unwrap!(spawner.spawn(north_task(p.PC13.into(), p.EXTI13.into())));
    unwrap!(spawner.spawn(north_task(p.PB14.into(), p.EXTI14.into())));
}
```
but the rustc compiler error with:
```
error[E0277]: the trait bound `embassy_stm32::exti::AnyChannel: From<EXTI13>` is not satisfied
  --> src/main.rs:38:62
   |
38 |     unwrap!(spawner.spawn(north_task(p.PC13.into(), p.EXTI13.into())));
   |                                                              ^^^^ the trait `From<EXTI13>` is not implemented for `embassy_stm32::exti::AnyChannel`, which is required by `EXTI13: Into<_>`
   |
   = note: required for `EXTI13` to implement `Into<embassy_stm32::exti::AnyChannel>`
```

I think the EXTI in STM32 HAL needs to implement an into() method to adapt to anychannel.

Dirbaio, 2024-06-02:
You can do it with [`.degrade()`](https://docs.embassy.dev/embassy-stm32/git/stm32f411ce/exti/trait.Channel.html#method.degrade) 

it's true this is inconsistent with gpio pins, which have both `.degrade()` and a `From/Into` impl. A pull request to fix this would be welcome if you're up for it :)

FourLeafTec, 2024-06-03:
`[.degrade()](https://docs.embassy.dev/embassy-stm32/git/stm32f411ce/exti/trait.Channel.html#method.degrade)` is ok,THX

closed 2024-06-03 by FourLeafTec