embassy-rs/embassy #3043: stm32 I2C implementation is different from document

lgjonathan, 2024-06-05:
While using the [stm32l011f4](https://docs.embassy.dev/embassy-stm32/git/stm32l011f4/i2c/struct.I2c.html) , I was trying to create a blocking I2C driver via `new_blocking` function as described in the doc.

However, the crate only allows me to create the Async version.
I looked into the the source code of "i2c" in my cargo registry and find there's no I2c<Blocking> in the code.
Is there a way to configure the embassy-stm32 so I can use the Blocking I2C?

Below is my Cargo.toml for reference
```toml
[dependencies]
embassy-stm32 = { version = "0.1.0", features = ["stm32l011f4", "time-driver-tim2", "memory-x"]  }
embassy-time = { version = "0.3.0", features = [] }
```

Is it because I have used multiple different STM32 on the same computer so there's some configuration issue? I did try `cargo clean` and rebuild it but to no avail.

Thanks for the your reply

Dirbaio, 2024-06-05:
docs on docs.embassy.dev are for embassy-stm32 from git main branch. see [this](https://embassy.dev/book/#_how_do_i_switch_to_the_main_branch) for how to switch.

in embassy-stm32 v0.1.0 from crates.io to create a blocking i2c you have to call `new()` and pass `NoDma, NoDma` as the DMA channels instead.

lgjonathan, 2024-06-07:
Thanks a lot for your prompt reply

closed 2024-06-07 by lgjonathan