embassy-rs/embassy #3046: [STM32] `uid::uid_hex()` always the same

shufps, 2024-06-05:
Hi,

this is a bit strange because the UID should be unique but I always get the same ID.

I'm using the UID for the serial number of my USB and I got `333347053438333411000000` on multiple STM32L072 :thinking: 

I tried it like this:
```rust
    // Create embassy-usb Config
    let mut config = embassy_usb::Config::new(0xc0de, 0xcafe);
    config.max_packet_size_0 = 64;
    config.manufacturer = Some("xxx");
    config.product = Some("yyy");
    config.serial_number = Some(uid::uid_hex());
```

I can see then a device:
```
/dev/serial/by-id/usb-xxx_yyy_333347053438333411000000-if000
```

but with different boards it's always the same.

What am I doing wrong? :see_no_evil:

shufps, 2024-06-06:
hmm, the memory address of the UID is correct in the PAC for the L072 :thinking:

shufps, 2024-06-10:
This works:
```
fn read_uid() {
    let base_address: usize = 0x1FF80050;

    let mut unique_id = [0u32;3];
    unsafe {
        unique_id = [
            ((base_address) as *const u32).read_volatile(),
            ((base_address+0x4) as *const u32).read_volatile(),
            ((base_address+0x14) as *const u32).read_volatile(),
        ];
    }
    info!("Unique ID: {:08X} {:08X} {:08X}", unique_id[0], unique_id[1], unique_id[2]);
}
```

probably there is an embassy bug because it probably assumes that all three 32bit words would follow directly after each other.

There even seems to be other inconsistencies like some STM32 having the 3rd word at offset `0x12`. The L072 has it on `0x14` :see_no_evil:

shufps, 2024-06-14:
This could be security relevant if UID is used as seed to initialize crypto things :thinking:

Dirbaio, 2024-06-16:
If you want this fixed, please send a PR instead of invoking the "could be security relevant" card.

shufps, 2024-06-19:
> If you want this fixed, please send a PR instead of invoking the "could be security relevant" card.

Ok, playing the "told you so" card somewhen in the future :rofl:  Just kidding.

Tbh, I can't fix it because it's above my skill level. I contributed a couple of other things but I guess it would require fundamental changes how the PAC is defined because it just has a base address and a "stride". But it's not prepared for reading from offsets like 0x00, 0x04 and 0x14.

So idk, can't do anything here :man_shrugging:

Dirbaio, 2024-06-21:
why close it? it's not fixed

hubpav, 2026-10-17:
Independent confirmation on an STM32L083CZ (embassy-stm32 0.6.0; the logic is the same on current `main`, which reads `UID.uid(idx)` for idx 0..3, i.e. a fixed 4-byte stride from the base).

RM0367 Rev 8, section 34.2 "Unique device ID registers (96 bits)": base address `0x1FF8_0050`, U_ID(31:0) at offset `0x00`, U_ID(63:32) at `0x04`, and U_ID(95:64) at **`0x14`**, not `0x08`.

Raw word reads on the chip:

```
+0x00 = 01473930   U_ID(31:0):  wafer 0x01 + lot "G90" (ASCII)
+0x04 = 36353931   U_ID(63:32): lot "6591" (ASCII)
+0x08 = 00000011   <- what uid() returns as the third word
+0x0c = ffffff00
+0x10 = 33a00011
+0x14 = 00340027   <- the documented U_ID(95:64)
+0x18 = 11000000
embassy uid() = [30, 39, 47, 01, 31, 39, 35, 36, 11, 00, 00, 00]
```

The `11 00 00 00` tail is the same `...11000000` as in the original report, so on L0 the third word is a factory constant and chips from the same wafer and lot get identical "unique" IDs.

A minimal fix would be to read the three words at offsets `[0x00, 0x04, 0x14]` on stm32l0 (and probably stm32l1, whose reference manual RM0038 documents the same split layout; not verified on hardware here) instead of the fixed stride, or to describe the three UID registers individually in stm32-data.

closed 2026-10-18 by xoviat