embassy-rs/embassy #3049: STM32H7 Sdmmc not functional

tycodeguy, 2024-06-05:
I need to do some logging on my STM32H7, and have tried everything I can to perform initialization and logging from both a Main and an embassy_executor task. I'm creating my peripheral with the new_4bit method, and when doing this, I lock up the chip when trying to call init_card(), or the card() reference grabber. If I simply try to write a block before doing this, I get an error from the returned result.

I'd appreciate help with any debugging steps or recommended fixes. 

Thanks!

JanekGraff, 2024-10-28:
Did you have any success with this? Facing a similar issue on a STM32L4 where `init_card()` hangs on sending the idle command ([here](https://github.com/embassy-rs/embassy/blob/80cb2bc4a5b8bfa424126dc7cf7c957c6894628b/embassy-stm32/src/sdmmc/mod.rs#L1077), specifically while waiting for the [command status](https://github.com/embassy-rs/embassy/blob/80cb2bc4a5b8bfa424126dc7cf7c957c6894628b/embassy-stm32/src/sdmmc/mod.rs#L999)

JanekGraff, 2024-10-28:
> Did you have any success with this? Facing a similar issue on a STM32L4 where `init_card()` hangs on sending the idle command [...]

Nvm, this was a clock configuration issue on my end. Seems the SDMMC Peripheral needs the HSI48 clock which i didn't initialize in my clock configuration.

ost-ing, 2025-01-30:
The example isn't working. Did anyone figure this one out for the stm32h7?