embassy-rs/embassy #3058: embassy-net does not have feature 'raw'.

Cam-2002, 2024-06-07:
Using a Debian environment with cargo & rustc versions 1.78.0.

Problem: Cargo cannot install crate `embassy-net` from crates.io with feature `raw`

Steps to reproduce problem:
1. Create new project from the example in `embassy/examples/rp/src/bin/wifi_tcp_server.rs`
2. Modify `Cargo.toml` to remove all paths for the Embassy crates, thus forcing cargo to install the crates from crates.io
3. Using `cargo build` results in the following error:
```
$ cargo build
error: failed to select a version for `embassy-net`.
    ... required by package `project v0.1.0 (/root/project)`
versions that meet the requirements `^0.4.0` (locked to 0.4.0) are: 0.4.0

the package `project` depends on `embassy-net`, with features: `raw` but `embassy-net` does not have these features.

failed to select a version for `embassy-net` which could resolve this conflict
```

Full `Cargo.toml` file:
```toml
[package]
edition = "2021"
name = "project"
version = "0.1.0"
license = "MIT OR Apache-2.0"

[dependencies]
embassy-embedded-hal = { version = "0.1.0", features = ["defmt"] }
embassy-sync = { version = "0.6.0", features = ["defmt"] }
embassy-executor = { version = "0.5.0", features = ["task-arena-size-32768", "arch-cortex-m", "executor-thread", "executor-interrupt", "defmt", "integrated-timers"] }
embassy-time = { version = "0.3.1", features = ["defmt", "defmt-timestamp-uptime"] }
embassy-rp = { version = "0.1.0", features = ["defmt", "unstable-pac", "time-driver", "critical-section-impl"] }
embassy-usb = { version = "0.2.0", features = ["defmt"] }
embassy-net = { version = "0.4.0", features = ["defmt", "tcp", "udp", "raw", "dhcpv4", "medium-ethernet"] }
embassy-net-wiznet = { version = "0.1.0", features = ["defmt"] }
embassy-futures = { version = "0.1.0" }
embassy-usb-logger = { version = "0.2.0" }
cyw43 = { version = "0.1.0", features = ["defmt", "firmware-logs"] }
cyw43-pio = { version = "0.1.0", features = ["defmt", "overclock"] }

defmt = "0.3"
defmt-rtt = "0.4"
fixed = "1.23.1"
fixed-macro = "1.2"

#cortex-m = { version = "0.7.6", features = ["critical-section-single-core"] }
cortex-m = { version = "0.7.6", features = ["inline-asm"] }
cortex-m-rt = "0.7.0"
critical-section = "1.1"
panic-probe = { version = "0.3", features = ["print-defmt"] }
display-interface-spi = "0.4.1"
embedded-graphics = "0.7.1"
st7789 = "0.6.1"
display-interface = "0.4.1"
byte-slice-cast = { version = "1.2.0", default-features = false }
smart-leds = "0.3.0"
heapless = "0.8"
usbd-hid = "0.7.0"

embedded-hal-1 = { package = "embedded-hal", version = "1.0" }
embedded-hal-async = "1.0"
embedded-hal-bus = { version = "0.1", features = ["async"] }
embedded-io-async = { version = "0.6.1", features = ["defmt-03"] }
embedded-storage = { version = "0.3" }
static_cell = "2"
portable-atomic = { version = "1.5", features = ["critical-section"] }
log = "0.4"
pio-proc = "0.2"
pio = "0.2.1"
rand = { version = "0.8.5", default-features = false }
embedded-sdmmc = "0.7.0"

[profile.release]
debug = 2
```

Dirbaio, 2024-06-07:
examples from latest git main work with embassy-net from latest git main.

if you want examples that work with the embassy-net v0.4.0 released on crates.io you have to go back in the repo history to the `embassy-net-v0.4.0` tag.

Cam-2002, 2024-06-08:
Forgive me, I thought I had seen 'raw' as a supported feature for embassy-net v0.4.0 on the docs.rs page, however upon checking again it appears I was mistaken. Will move forward with the example from the embassy-net-v0.4.0 tag in the repo history.

Dirbaio, 2025-01-10:
embassy-net v0.6.0 is now released.

closed 2025-01-10 by Dirbaio