embassy-rs/embassy #3070: Getting started with nrf 52840 Error: No debug probes were found.

Gibbz, 2024-06-13:
I'm trying to get the examples up and running on my nrf 52840 under linux (arch/manjaro). 

Ive setup the udev rules from https://github.com/NordicSemiconductor/nrf-udev

If i list usb devices its there  'ls /dev/ttyACM*' shows: '/dev/ttyACM0'

But when i list the devices using 'probe-rs list' there are none found: 'Error: No debug probes were found.'

How do I get the device to show?

lulf, 2024-06-13:
You might want to check your USB cable/try a different one if you've got. Some are charging only.

Dirbaio, 2024-06-13:
What nRF board do you have? You need it either to have a builtin debug probe, or you need a separate debug probe.

Gibbz, 2024-06-17:
I have a cheap clone board (Promicro Adafruit nice!nano) found here: https://www.aliexpress.com/item/1005006446457448.html?spm=a2g0o.order_list.order_list_main.50.7a051802a6u8xp

Ive figured out i double tap the reset button to give me access to a mountable partition which accepts a uf2 file for flashing.

What is a decent debug probe you guys use?

Dirbaio, 2024-06-17:
the uf2 thing is a builtin bootloader, it's not a debug probe. It doesn't work with probe-rs. You can try using it with Rust with [elf2uf2-rs](https://github.com/JoNil/elf2uf2-rs) but I really don't recommend it: you won't get logs especially from panic messages, so if your firmware crashes you won't be able to see what's wrong and fix it.

for an actual debug probe i'd recommend either the Raspberry Pi Debug probe ($15) or a Raspberry Pi Pico ($5) with the [debugprobe](https://github.com/raspberrypi/debugprobe) firmware. (Both are exactly the same, just the Debug Probe has a more convenient form factor).

You'll have to solder to the DIO, CLK pads on the back.

Gibbz, 2024-06-18:
Thanks! I've ordered a probe. I figure it's easier to work with in the long run 👍

8cichy8, 2024-06-19:
I tried this board too, but I'm not expert in soldering, so for me those pads on the back was too close to each other. So in the end I bought another board with similar form factor, but with holes for DIO and CLK (It was about 3$ more for that board - 52840nano).

For UF2 conversion I did used uf2conv.py from https://github.com/microsoft/uf2/tree/master/utils . Something like this:
cargo build && arm-none-eabi-objcopy --output-target binary ./target/thumbv7em-none-eabihf/debug/blink ./target/thumbv7em-none-eabihf/debug/blink.bin && ./uf2conv.py ./target/thumbv7em-none-eabihf/debug/blink.bin -o ./target/thumbv7em-none-eabihf/debug/blink.uf2 -c -b 0x26000 -f 0xADA52840

With "FLASH : ORIGIN = 0x00026000, LENGTH = 872K" in memory.x, because there is Softdevices S140 6.1.1.

Little info about this board: https://github.com/joric/nrfmicro/wiki/Alternatives#supermini-nrf52840

Gibbz, 2024-06-25:
Okay ive got the rpi probe up and running! Easy thanks for help guys!

To solder the small pads I did the following:
- heat the pad with soldering iron, put a blob of solder on the pad
- tin the wires with solder
- reheat the pad, and insert the wire into the blob, remove the soldering iron, hold till cool
- done!

closed 2024-06-25 by Gibbz

0xorial, 2024-12-07:
could someone please provide more explanations of how to get probe-rs working? I soldered the pads, and if i connect I get:
```
user@Mac Downloads % ./nrfutil device list
2120000
product         USB Serial
ports           /dev/tty.usbserial-2120
traits          serialPorts, usb

Found 1 supported device(s)

user@Mac Downloads % probe-rs list        
No debug probes were found.
```

which is same as when I simply connect it to usb.