embassy-rs/embassy #3076: Examples should have a "different chip" checklist

jamesmunns, 2024-06-14:
It would be good to have a checklist in the source file of every example, to ensure people trying to use examples for different chips/boards know what they need to do. This is a VERY common question in chat, which means likely even more people who never show up in chat have the same issue.

I think having this in-code would be very helpful (from the stm32f4 blinky example):

```rust
// This example is written for the stm32f429-nucleo, with a stm32f429zi chip.
//
// If you are using a different board or chip, make sure you update the following:
//
// * [ ] Update .cargo/config.toml with the correct `probe-rs run --chip STM32F429ZITx` chip name
// * [ ] Update Cargo.toml to have the correct `embassy-stm32` feature, it is currently `stm32f429zi`
// * [ ] If your board has a special clock or power configuration, make sure that it is set up
//       appropriately. If you are unsure, please drop by the Embassy Matrix chat for support.
let p = embassy_stm32::init(Default::default());
info!("Hello World!");
```

This is a bit tedious to write out, but I think it will be helpful to reduce the amount of frustration from new users.

jamesmunns, 2024-06-14:
If we do want to do this, here's a tracking list of all example kinds that need this:

* [ ] boot
* [ ] nrf-rtos-trace
* [ ] nrf51
* [ ] nrf52810
* [ ] nrf52840
* [ ] nrf52840-rtic
* [ ] nrf5340
* [ ] nrf9160
* [ ] rp
* [ ] std
* [ ] stm32c0
* [ ] stm32f0
* [ ] stm32f1
* [ ] stm32f2
* [ ] stm32f3
* [ ] stm32f334
* [x] stm32f4 - #3077 
* [ ] stm32f469
* [ ] stm32f7
* [ ] stm32g0
* [ ] stm32g4
* [ ] stm32h5
* [ ] stm32h7
* [ ] stm32h7rs
* [ ] stm32l0
* [ ] stm32l1
* [ ] stm32l4
* [ ] stm32l5
* [ ] stm32u0
* [ ] stm32u5
* [ ] stm32wb
* [ ] stm32wba
* [ ] stm32wl
* [ ] wasm

jedkutai, 2026-08-27:
I’m interested in picking this up. I saw that #3077 was closed during the old PR cleanup. Would you still like this implemented, and is the preferred approach now a short note in each example linking to centralized documentation?

Dirbaio, 2026-08-27:
I think the examples are growing out of control. There's a huge amount of copypaste, doing API changes is a pain because you have to update the 20 copies of the same example.

Maybe what we want instead is a "project generator"? you pick your chip/board then you get a list of all examples applicable to it (blinky, spi, i2c, uart, network, wifi, adc, whatever). They would be "templated" somehow so you can generate, say, an i2c example for your exact weird stm32 chip.

This is an idea i've had on the back of my mind for quite a long time but never found the time to sit down and execute it...

jedkutai, 2026-08-27:
That sounds interesting, and I'd be interested in exploring it.

Would a small proof of concept be useful before trying to design the full system? I was thinking of starting with STM32 and supporting just a couple of examples, such as blinky and I2C, to figure out what the templating and chip selection could look like.

If that sounds reasonable, is there any existing infrastructure or chip/board metadata in Embassy that you would want the generator to build on?

Dirbaio, 2026-08-27:
stm32 gets metadata from https://github.com/embassy-rs/stm32-data/
there's nothing that attempts to build one metadata for all chips from different vendors. I'm not sure if that's feasible / worth trying.