embassy-rs/embassy #3078: macro expansion broken on latest nightly (2024-06-13)

Usioumeo, 2024-06-14:
I am developing a RISCV project based on embassy and ch32-hal, I have update my machine and installed the latest nightly version. Unfortunately the embassy macro expansion seems to have broken, and the error is not that helpful.
Now i get this error:
```rust
error: item does not constrain `__embassy_main::Fut::{opaque#0}`, but has it in its signature
  --> src/bin/main.rs:59:1
   |
59 | #[embassy_executor::main(entry = "qingke_rt::entry")]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ in this procedural macro expansion
   |
  ::: <.cargo path>/registry/src/index.crates.io-6f17d22bba15001f/embassy-executor-macros-0.4.1/src/lib.rs:58:1
   |
58 | pub fn task(args: TokenStream, item: TokenStream) -> TokenStream {
   | ---------------------------------------------------------------- in this expansion of `#[::embassy_executor::task]`
   |
   = note: consider moving the opaque type's declaration and defining uses into a separate module
```
Having glazed at the changelog [here](https://doc.rust-lang.org/nightly/releases.html) it seems they have worked on impl traits and opaque types... In the weekend I could have some time to check on the issue... Let me know

joelpmichael, 2024-06-18:
+1, on same embedded platform (if that helps), on a brand new project created from the ch32-hal-template at https://github.com/ch32-rs/ch32-hal-template

my `rust-toolchain.toml` contains:

```
[toolchain]
targets = ["riscv32imac-unknown-none-elf"]
channel = "nightly-2024-06-12" # works
#channel = "nightly-2024-06-13" # broken
```

I haven't yet looked into what language change may have caused this.

Dirbaio, 2024-06-18:
this broke due to https://github.com/rust-lang/rust/issues/117861 https://github.com/rust-lang/rust/pull/113169

it's fixed in git main since https://github.com/embassy-rs/embassy/pull/2574 so i'm going to close this. Next crates.io release will include the fix.

closed 2024-06-18 by Dirbaio