embassy-rs/embassy #3096: Help wanted: embassy-net-ppp

KorribanMaster, 2024-06-19:
Hello,
I'm trying to set up a project with a simcom sim868 GSM/GNSS module using PPP. I managed to setup the module using PPP on linux successfully using the following [guide](https://andino.systems/extensions/2g-modem-sim800l/ppp) as well as talking to the simcom module via serial from an stm32f449 and connecting it to a network and sending data using AT commands but using the ppp network from embassy seems much nicer integrated. 

However I can not figure out how to setup embassy ppp from the [example](https://github.com/embassy-rs/embassy/blob/b0172bb58217d625a13fed8122827b8d0b03c46a/examples/std/src/bin/net_ppp.rs) to connect using the real modem. I guess what i am missing is the modem and carrier specifics that are setup via the ppp config and the chatscript when using linux and pppd. 

Has anyone already setup a real gsm modem using ppp and maybe could provide an example or some cues on the process?

Dirbaio, 2024-06-19:
embassy-net-ppp only does the "PPP" part, not the "AT command" part. You can use it with a modem but you have to do the AT commands yourself. 

Run the AT commands needed to setup the modem (write it as a string to the uart, then wait the modem to reply with OK). The last command `ATD*99#` is special: write it as a string, then wait for the modem to reply with CONNECT. After this, the UART is switched to "data mode", speaking PPP. After this is done, you can start forwarding bytes between the serial port and embassy-net-ppp.

overflowz, 2024-08-22:
> Hello, I'm trying to set up a project with a simcom sim868 GSM/GNSS module using PPP. I managed to setup the module using PPP on linux successfully using the following [...]

I'm actually working on this and managed to make it work (dirty code, a lot of messing). I can provide you the steps what you should look for (right now I have huge mess in code, doing a lot of this and that and mixed up together). I did exactly what Dirbaio said and worked pretty well.

1. you need to setup two buffers (rx and tx), like `[u8; N_RX/TX];`
2. connect to modem via UART, dirty example taken from my code:
```rs
    // somewhere top before main
    bind_interrupts!(struct UartIrqs {
        USART1 => embassy_stm32::usart::BufferedInterruptHandler<peripherals::USART1>;
    });

    // in main
     let mut uart = embassy_stm32::usart::BufferedUart::new(
         p.USART1,
         UartIrqs,
         p.PA10,
         p.PA9,
         tx_buffer,
         rx_buffer,
         Default::default(),
     )
     .unwrap();
```
3. send commands to modem to setup and switch to data mode (in my case I send `AT+CGDCONT=1,"ip","apn"\r` followed by `ATD*99#\r`. example sending / receiving part:
```rs
    // reading & parsing output is up to you, to make sure modem is up, got successful replies on commands, etc.
    uart.write_all("AT\r".as_bytes()).await.unwrap();
    uart.flush().await.unwrap();
    Timer::after_secs(1).await; // wait for a bit for modem to respond
    uart.read(&mut buf).await.unwrap();
```
4. now comes the embassy_net_ppp part:
```rs
let state = embassy_net_ppp::State::<N_RX, N_TX>::new();
let (device, runner) = embassy_net_ppp::new(state);

let cfg = embassy_net_ppp::Config {
  username: "foo".as_bytes(),
  password: "bar".as_bytes(),
}

runner.run(uart, cfg, |stat| {
  info!("stat: {}", stat);
});
```

then basically follow the docs, you already have the connection. create network stack, etc.

KorribanMaster, 2024-09-09:
ah i was only missing the `ATD*99#\r` part i already figured i probably need to connect the modem first. Thanks a lot to both of you. Maybe i can make a PR to the README of embassy_net_ppp to explain this a little better

closed 2024-10-04 by KorribanMaster