embassy-rs/embassy #3097: Cannot create ospi instance for stm32h7b0

HaoboGu, 2024-06-19:
I'm trying to use stm32h7b0's octospi peripheral as the following:

```rust
#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    let ospi_config = ospi::Config::default();

    let ospi = Ospi::new_quadspi(
        p.OCTOSPI1,
        p.PA3,
        p.PC9,
        p.PD12,
        p.PE2,
        p.PD13,
        p.PB6,
        p.MDMA,
        ospi_config,
    );
}
```

But the compiler reports all the pins are not impl for `OCTOSPI1`. I checked the `stm32-data-generated`, it seems that the pins are defined in `OCTOSPIM` peripheral, but not `OCTOSPI1` and `OCTOSPI2`:

https://github.com/embassy-rs/stm32-data-generated/blob/e30c7694016fa9a326f153fff84dedb0c02b8676/data/chips/STM32H7B0VB.json#L4147-L4235

However, `Ospi::new_quadspi` doesn't accept `p.OCTOSPIM` as the `peri`. Is there any method to fix it?

---- Attached: parts of the compiler error
```
error[E0277]: the trait bound `PA3: embassy_stm32::ospi::SckPin<embassy_stm32::peripherals::OCTOSPI1>` is not satisfied
   --> src/main.rs:114:20
    |
114 |       let mut ospi = Ospi::new_quadspi(
    |  ____________________^
115 | |         p.OCTOSPI1,
116 | |         p.PA3,
117 | |         p.PC9,
...   |
123 | |         ospi_config,
124 | |     );
    | |_____^ the trait `embassy_stm32::ospi::SckPin<embassy_stm32::peripherals::OCTOSPI1>` is not implemented for `PA3`
    |
note: required by a bound in `Ospi::<'d, T, embassy_stm32::mode::Async>::new_quadspi`
   --> C:\Users\haobo\.cargo\git\checkouts\embassy-9312dcb0ed774b29\b0172bb\embassy-stm32\src\ospi\mod.rs:785:39
    |
783 |     pub fn new_quadspi(
    |            ----------- required by a bound in this associated function
784 |         peri: impl Peripheral<P = T> + 'd,
785 |         sck: impl Peripheral<P = impl SckPin<T>> + 'd,
    |                                       ^^^^^^^^^ required by this bound in `Ospi::<'d, T, Async>::new_quadspi`

error[E0277]: the trait bound `PC9: embassy_stm32::ospi::D0Pin<embassy_stm32::peripherals::OCTOSPI1>` is not satisfied
   --> src/main.rs:114:20
    |
114 |       let mut ospi = Ospi::new_quadspi(
    |  ____________________^
115 | |         p.OCTOSPI1,
116 | |         p.PA3,
117 | |         p.PC9,
...   |
123 | |         ospi_config,
124 | |     );
    | |_____^ the trait `embassy_stm32::ospi::D0Pin<embassy_stm32::peripherals::OCTOSPI1>` is not implemented for `PC9`
    |
note: required by a bound in `Ospi::<'d, T, embassy_stm32::mode::Async>::new_quadspi`
   --> C:\Users\haobo\.cargo\git\checkouts\embassy-9312dcb0ed774b29\b0172bb\embassy-stm32\src\ospi\mod.rs:786:38
    |
783 |     pub fn new_quadspi(
    |            ----------- required by a bound in this associated function
...
786 |         d0: impl Peripheral<P = impl D0Pin<T>> + 'd,
    |                                      ^^^^^^^^ required by this bound in `Ospi::<'d, T, Async>::new_quadspi`

error[E0277]: the trait bound `PD12: embassy_stm32::ospi::D1Pin<embassy_stm32::peripherals::OCTOSPI1>` is not satisfied
   --> src/main.rs:114:20
    |
114 |       let mut ospi = Ospi::new_quadspi(
    |  ____________________^
115 | |         p.OCTOSPI1,
116 | |         p.PA3,
117 | |         p.PC9,
...   |
123 | |         ospi_config,
124 | |     );
    | |_____^ the trait `embassy_stm32::ospi::D1Pin<embassy_stm32::peripherals::OCTOSPI1>` is not implemented for `PD12`
    |
note: required by a bound in `Ospi::<'d, T, embassy_stm32::mode::Async>::new_quadspi`
   --> C:\Users\haobo\.cargo\git\checkouts\embassy-9312dcb0ed774b29\b0172bb\embassy-stm32\src\ospi\mod.rs:787:38
```

Dirbaio, 2024-06-19:
OCTOSPIM is a new peripheral that sits on top of OSPI1 and OSPI2 and does some kind of pin muxing. So, fixing this requires adding support in embassy-stm32 for configuring the OCTOSPIM I think.

HaoboGu, 2024-06-19:
> OCTOSPIM is a new peripheral that sits on top of OSPI1 and OSPI2 and does some kind of pin muxing. So, fixing this requires adding support in embassy-stm32 for configuring the OCTOSPIM I think.

Is there any reference about doing this type of configuring? I'd like to fix it if possible

Dirbaio, 2024-06-19:
check the Reference Manual for your chip, it explains how OCTOSPIM works.

HaoboGu, 2024-06-19:
> check the Reference Manual for your chip, it explains how OCTOSPIM works.

Should I implement a separate peripheral `ospim`, or just update current `ospi`?

closed 2024-10-13 by Dirbaio