embassy-rs/embassy #3108: Future for Timer Update Interrupt missing

shufps, 2024-06-22:
Hi,

I'm trying to get a DDS to work but currently there seems not to be a way to have a future waiting for the timer update interrupt or timer compare interrupt

Unfortunately this is needed because DDS is completly different to waveforms that are played back from RAM via DMA.

It requires to calculate the next PWM value based on a fixed point phase accumulator pointing to a sine table but this requires to update the PWM value after each timer update interrupt - better would be timer compare interrupt.

I saw in the low_level.rs that there is some function that can return if a timer interrupt happened but it would be a "bruteforce"-polling method that wastes a lot of CPU power. Better would be a Future that `await`s in some task.

I can contribute some DDS sample if someone could point me in the right direction.

Thank you very much,
Thomas

shufps, 2024-06-22:
Mmhmm maybe I can C&P parts from the Input Interrupt :thinking:

shufps, 2024-06-23:
I think this should work with minimum overhead:
```rust
#[interrupt]
fn TIM2() {
    unsafe {
        // get next value of DDS
        DDS_AKKU = DDS_AKKU.wrapping_add(DDS_INCR);
        let value = DDS_SINE_DATA[(DDS_AKKU >> 24) as usize];
        pac::TIM2.ccr(3).modify(|w| w.set_ccr(value as u16));
    }
}
```

romainreignier, 2024-06-29:
I don't know what do you mean by DDS but this PR might interest you for the futures: https://github.com/embassy-rs/embassy/pull/3029#issuecomment-2144021658

shufps, 2024-06-30:
> I don't know what do you mean by DDS

DDS means "direct digital synthesis" - it's like the opposite of digitizing an analog audio into digital samles.

DDS creates digital samples of a sine wave that is then converted to analog external.

The trick on DDS is that you can generate arbitrary sine frequencies - like you can digitize arbitrary sine wave frequencies (ofc with respect to Nyquist) of an analogue signal.

It's a little bit of an edge case, not so much used often, but it's needed for digital signal processing often.

Oh nice to see that I'm not the only one who would need Futures for Timers :raised_hands: