embassy-rs/embassy #3112: [embassy-net]: Fails to recover if sockets are open while link is going down

MathiasKoch, 2024-06-24:
Hi!

When a `smoltcp` link is brought down, without closing its sockets first, what is supposed to hinder this piece of code from executing very rapidly? (https://github.com/embassy-rs/embassy/blob/d5badb94d2098b0360ed1ab8015c37f819771445/embassy-net/src/lib.rs#L833-L859)

This would happen until the socket timeout is reached, and if using the `embedded-nal` abstractions, it is not possible to set a timeout, which means that it ends up in a locked infinite loop, that it doesn't even seem to be able to recover from if the link comes back up? 

With the result of

```
└─ smoltcp::iface::interface::{impl#1}::socket_egress::{closure#1}::{closure#0} @ /home/mathias/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
284.524019 DEBUG failed to transmit IP: device exhausted
└─ smoltcp::iface::interface::{impl#1}::socket_egress::{closure#1}::{closure#0} @ /home/mathias/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
284.524074 DEBUG failed to transmit IP: device exhausted
└─ smoltcp::iface::interface::{impl#1}::socket_egress::{closure#1}::{closure#0} @ /home/mathias/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
284.524129 DEBUG failed to transmit IP: device exhausted
└─ smoltcp::iface::interface::{impl#1}::socket_egress::{closure#1}::{closure#0} @ /home/mathias/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
284.524185 DEBUG failed to transmit IP: device exhausted
└─ smoltcp::iface::interface::{impl#1}::socket_egress::{closure#1}::{closure#0} @ /home/mathias/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
284.524241 DEBUG failed to transmit IP: device exhausted
└─ smoltcp::iface::interface::{impl#1}::socket_egress::{closure#1}::{closure#0} @ /home/mathias/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
284.524296 DEBUG failed to transmit IP: device exhausted
└─ smoltcp::iface::interface::{impl#1}::socket_egress::{closure#1}::{closure#0} @ /home/mathias/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
284.524351 DEBUG failed to transmit IP: device exhausted
└─ smoltcp::iface::interface::{impl#1}::socket_egress::{closure#1}::{closure#0} @ /home/mathias/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
284.524407 DEBUG failed to transmit IP: device exhausted
└─ smoltcp::iface::interface::{impl#1}::socket_egress::{closure#1}::{closure#0} @ /home/mathias/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
284.524463 DEBUG failed to transmit IP: device exhausted
└─ smoltcp::iface::interface::{impl#1}::socket_egress::{closure#1}::{closure#0} @ /home/mathias/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
284.524520 DEBUG failed to transmit IP: device exhausted
└─ smoltcp::iface::interface::{impl#1}::socket_egress::{closure#1}::{closure#0} @ /home/mathias/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
284.524575 DEBUG failed to transmit IP: device exhausted
└─ smoltcp::iface::interface::{impl#1}::socket_egress::{closure#1}::{closure#0} @ /home/mathias/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
284.524631 DEBUG failed to transmit IP: device exhausted
└─ smoltcp::iface::interface::{impl#1}::socket_egress::{closure#1}::{closure#0} @ /home/mathias/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
284.524687 DEBUG failed to transmit IP: device exhausted
└─ smoltcp::iface::interface::{impl#1}::socket_egress::{closure#1}::{closure#0} @ /home/mathias/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
284.524742 DEBUG failed to transmit IP: device exhausted
└─ smoltcp::iface::interface::{impl#1}::socket_egress::{closure#1}::{closure#0} @ /home/mathias/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
284.524799 DEBUG failed to transmit IP: device exhausted
└─ smoltcp::iface::interface::{impl#1}::socket_egress::{closure#1}::{closure#0} @ /home/mathias/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18 
```

ProfFan, 2024-10-18:
Same issue here, if the other end is not available the sending would fail and `failed to transmit IP: device exhausted` spams the entire console...

ProfFan, 2024-10-18:
Ok, I think my problem is not flushing `embassy-net`'s transmit queues. The solution is simple: keep the `Runner` running 🤷

Luctins, 2024-10-29:
I'm having the same message here. For me the issue is that I don´t flush my sockets I get very variable transfer speeds and a flood of these messages occasionally but if after every write I flush the socket the transfer rate is more stable and there are no errors logged.

nikvoid, 2024-12-02:
I am using embassy-net-ppp and when I have to switch modem into command mode, data link must be suspended, and this flood starts if it will be suspended for too long. Also if PPP connection terminates sockets won't disconnect and will just sit in infinite loop spamming this message.