embassy-rs/embassy #3122: Suddenly getting error with no changes....

Gibbz, 2024-06-26:
My embassy project has been working fine since i set it up, but suddenly I started getting the following error:
```
error: `#[panic_handler]` function required, but not found
```

Ive tried adding to the top of main.rs
```
#[panic_handler] fn panic(_info: &core::panic::PanicInfo) -> ! { loop {} }
```

But then i get another error saying 
```
`panic_handler` is ambiguous
ambiguous because of a name conflict with a builtin attribute
use `crate::panic_handler` to refer to this attribute macro unambiguously
```

Has anyone had this issue suddenly pop up?
What do I do to fix this?

8cichy8, 2024-06-26:
> error: `#[panic_handler]` function required, but not found
This pops up for me, if I don't have imported:
`use {defmt_rtt as _, panic_probe as _};`
But if you haven't changed anything ... maybe `cargo clean` and rebuild will help

Gibbz, 2024-06-27:
Thanks!
```use {defmt_rtt as _, panic_probe as _}; ```
was the solution, it must have been removed somehow!

closed 2024-06-27 by Gibbz