embassy-rs/embassy #3147: shared i2c bus help

Gibbz, 2024-07-03:
Im trying to setup a shared i2c bus on my nrf52840. Im not having much luck with the example code:
https://docs.embassy.dev/embassy-embedded-hal/git/default/shared_bus/asynch/i2c/index.html

My source:
```rust

// external imports
use embassy_nrf::{gpio::Pin, peripherals::TWISPI0};
use embassy_time::Timer;
use embassy_executor::Spawner;
use defmt::*;
use {defmt_rtt as _, panic_probe as _};

// Static i2c/twi mutex for shared-bus functionality
use static_cell::StaticCell;
use embassy_sync::mutex::Mutex;
use embassy_nrf::twim::{self, Twim};
use embassy_embedded_hal::shared_bus::asynch::i2c::I2cDevice;
use embassy_sync::blocking_mutex::raw::NoopRawMutex;

static I2C_BUS: StaticCell<Mutex<NoopRawMutex, Twim<TWISPI0>>> = StaticCell::new();

#[embassy_executor::main]
async fn main(spawner: Spawner) {
    let p = embassy_nrf::init(Default::default());

    // DEBUG: add sleep incase we need to flash during debug and get a crash
    Timer::after_secs(2).await;

    // https://docs.embassy.dev/embassy-embedded-hal/git/default/shared_bus/asynch/i2c/index.html

    let i2c_bus = {
        use embassy_nrf::{bind_interrupts, peripherals::{self}};
        bind_interrupts!(struct Irqs {
            SPIM0_SPIS0_TWIM0_TWIS0_SPI0_TWI0 => twim::InterruptHandler<peripherals::TWISPI0>;
        });
        let config = twim::Config::default();
        let i2c = Twim::new(p.TWISPI0, Irqs, p.P0_03, p.P0_04, config);
        let i2c_bus = Mutex::new(i2c);
        I2C_BUS.init(i2c_bus)
    };

    let i2c_dev1 = I2cDevice::new(i2c_bus);
}

```

Dirbaio, 2024-07-03:
in Cargo.toml, try changing embassy-sync from v0.6.0 to v0.5.0.

the released `embassy-embedded-hal` crate is still on v0.5.0.

Gibbz, 2024-07-03:
Thanks, that's fixed that issue. Now I have an issue with the i2c device.

from main.rs
```
static I2C_BUS: StaticCell<Mutex<NoopRawMutex, Twim<TWISPI0>>> = StaticCell::new();

#[embassy_executor::main]
async fn main(spawner: Spawner) {
    let p = embassy_nrf::init(Default::default());

    let i2c_bus = {
        use embassy_nrf::{bind_interrupts, peripherals::{self}};
        bind_interrupts!(struct Irqs {SPIM0_SPIS0_TWIM0_TWIS0_SPI0_TWI0 => twim::InterruptHandler<peripherals::TWISPI0>;});
        let config = twim::Config::default();
        let i2c = Twim::new(p.TWISPI0, Irqs, p.P0_06, p.P0_08, config); // sda: p0.06, scl: p0.08
        let i2c_bus = Mutex::new(i2c);
        //let i2c_bus = NoopMutex::new(RefCell::new(i2c));
        //let i2c_bus = embassy_sync::mutex::Mutex::new(RefCell::new(i2c));
        I2C_BUS.init(i2c_bus)
    };

    let i2c_dev1 = I2cDevice::new(i2c_bus);
    let i2c_dev2 = I2cDevice::new(i2c_bus);

    // Throttle ADC (output)
    use crate::device_throttle_dac::dac;
    spawner.must_spawn(dac(
        i2c_dev1
    ));
  }
```

and the device_throttle_dac.rs:
```
use embassy_embedded_hal::shared_bus::asynch::i2c::I2cDevice;
use embassy_nrf::{peripherals::TWISPI0, twim::Twim};
use defmt::*;
use embassy_sync::blocking_mutex::raw::NoopRawMutex;
use mcp4725::MCP4725;

static DEVICE_ID : &str = "THROTTLE DAC";

#[embassy_executor::task]
pub async fn dac (
    i2c: I2cDevice<'static, NoopRawMutex, Twim<'static, TWISPI0>>
) {
    let address = 0x60;
    let mut dac = MCP4725::new(i2c, address);
    let _ = dac.set_dac_and_eeprom(mcp4725::PowerDown::Normal, 0); // set 0 volts output

    info!("{} : Entering main loop", DEVICE_ID);
    loop {
        let read = dac.read().unwrap();
        info!("{}", read.data());
    }
}
```

main.rs builds fine, but in the device ```let mut dac = MCP4725::new(i2c, address);``` line gives errors:
```
error[E0277]: the trait bound `embassy_embedded_hal::shared_bus::asynch::i2c::I2cDevice<'_, NoopRawMutex, Twim<'static, TWISPI0>>: embedded_hal::blocking::i2c::Read` is not satisfied
  --> src/device_throttle_dac.rs:16:19
   |
16 |     let mut dac = MCP4725::new(i2c, address);
   |                   ^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `embedded_hal::blocking::i2c::Read` is not implemented for `embassy_embedded_hal::shared_bus::asynch::i2c::I2cDevice<'_, NoopRawMutex, Twim<'static, TWISPI0>>`
   |
   = help: the following other types implement trait `embedded_hal::blocking::i2c::Read<A>`:
             Twim<'a, T>
             embassy_embedded_hal::shared_bus::blocking::i2c::I2cDevice<'_, M, BUS>
note: required by a bound in `MCP4725`
  --> /home/bronson/.cargo/registry/src/index.crates.io-6f17d22bba15001f/mcp4725-0.4.2/src/lib.rs:60:10
   |
58 | pub struct MCP4725<I2C>
   |            ------- required by a bound in this struct
59 | where
60 |     I2C: Read + Write,
   |          ^^^^ required by this bound in `MCP4725`

```

and

```
error[E0277]: the trait bound `embassy_embedded_hal::shared_bus::asynch::i2c::I2cDevice<'_, NoopRawMutex, Twim<'static, TWISPI0>>: embedded_hal::blocking::i2c::Write` is not satisfied
  --> src/device_throttle_dac.rs:16:19
   |
16 |     let mut dac = MCP4725::new(i2c, address);
   |                   ^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `embedded_hal::blocking::i2c::Write` is not implemented for `embassy_embedded_hal::shared_bus::asynch::i2c::I2cDevice<'_, NoopRawMutex, Twim<'static, TWISPI0>>`
   |
   = help: the following other types implement trait `embedded_hal::blocking::i2c::Write<A>`:
             Twim<'a, T>
             embassy_embedded_hal::shared_bus::blocking::i2c::I2cDevice<'_, M, BUS>
note: required by a bound in `MCP4725`
  --> /home/bronson/.cargo/registry/src/index.crates.io-6f17d22bba15001f/mcp4725-0.4.2/src/lib.rs:60:17
   |
58 | pub struct MCP4725<I2C>
   |            ------- required by a bound in this struct
59 | where
60 |     I2C: Read + Write,
   |                 ^^^^^ required by this bound in `MCP4725`

```

lulf, 2024-07-03:
Could you list the embassy crate versions you're using and any entries in the [patch.crates-io] section?

Gibbz, 2024-07-03:
My cargo.toml:
```
[package]
edition = "2021"
name = "bike-aid-nrf-rs"
version = "0.1.0"
authors = 

[features] 
default = [
    "ble-l2cap",
    "ble-gatt-server",
    "ble-gatt-client",
    "ble-sec",
]

ble-l2cap = ["nrf-softdevice/ble-l2cap"]
ble-gatt-server = ["nrf-softdevice/ble-gatt-server"]
ble-gatt-client = ["nrf-softdevice/ble-gatt-client"]
ble-sec = ["nrf-softdevice/ble-sec"]

nrf52840 = [
  "embassy-nrf/nrf52840",
  "nrf-softdevice/nrf52840",
  "nrf-softdevice/s140",
  "dep:nrf-softdevice-s140"
]

[dependencies]
# embassy
embassy-futures = { version = "0.1.0"}
embassy-sync = { version = "0.5.0", features = ["defmt"] }
embassy-executor = { version = "0.5.0", features = ["task-arena-size-32768", "arch-cortex-m", "executor-thread", "executor-interrupt", "defmt", "integrated-timers"] }
embassy-time = { version = "0.3.1", features = ["defmt", "defmt-timestamp-uptime"] }
embassy-nrf = { version = "0.1.0", features = ["defmt", "nrf52840", "time-driver-rtc1", "gpiote", "unstable-pac", "time"] }
embassy-embedded-hal = "0.1.0"

# log
defmt = "0.3.8"
defmt-rtt = "0.4.1"

# arm
cortex-m = { version = "0.7.7", features = ["inline-asm"] } # , "critical-section-single-core"
cortex-m-rt = "0.7.3"

# misc
panic-probe = { version = "0.3.1", features = ["print-defmt"] }
serde = { version = "1.0.136", default-features = false }

# device driver
mcp4725 = "0.4.2"
ads1x1x = "0.2.2"

# maths
num-traits = { version = "0.2.19", default-features = false, features = ["libm"] }

# nrf softdevice bluetooth
nrf-softdevice-s140 = { version = "0.1.2", optional = true }
nrf-softdevice = { version = "0.1.0", features = ["nrf52840", "s140", "defmt", "ble-peripheral", "ble-central", "critical-section-impl"] }
embedded-storage = "0.3.1"
embedded-storage-async = "0.4.1"
futures = { version = "0.3.29", default-features = false }
fixed = "1.24.0"
heapless = "0.8.0"
atomic-pool = "1.0.1"
static_cell = "2.0.0"

[profile.release]
debug = 2

```

lulf, 2024-07-04:
> 
> ```
> use embassy_embedded_hal::shared_bus::asynch::i2c::I2cDevice;
>

This is the issue. You're trying to use the asynch::i2c::I2cDevice instead of the blocking::i2c::I2cDevice. The driver requires
the blocking variant, so you need to import `embassy_embedded_hal::shared_bus::blocking::i2c::I2cDevice` in your code.

Gibbz, 2024-07-04:
Awesome! That was the problem I got it working! Thanks!

If anyone's looking for the source code...

Setup the i2c shared bus in main:
https://github.com/bit-shift-io/bike-aid/blob/main/bike-aid-nrf-rs/src/main.rs

Example tasks:
https://github.com/bit-shift-io/bike-aid/blob/main/bike-aid-nrf-rs/src/device_throttle_adc.rs
https://github.com/bit-shift-io/bike-aid/blob/main/bike-aid-nrf-rs/src/device_throttle_dac.rs