embassy-rs/embassy #3164: Reading Information Elements (IE) with CYW43

1fabi0, 2024-07-10:
The `BSSInfo` already provides a offset and length of the IE (see [here](https://github.com/embassy-rs/embassy/blob/main/cyw43/src/structs.rs#L527)). But as far as I see this there is currently no way of reading the IE. IE can for example provide information if the Service Set requires Authentication and which Authentication is required. 

Am I right and implementation of reading IE values is missing, or is there a way to read IE?

Dirbaio, 2024-07-10:
it's not implemented, PRs welcome!

Frostie314159, 2024-08-16:
This is the relevant code in the C driver https://github.com/Infineon/wifi-host-driver/blob/9e4da8f0a5f5bf480d5e3661d028f68877423beb/WHD/COMPONENT_WIFI5/src/whd_wifi_api.c#L2413C44-L2413C45

plugwash, 2025-03-07:
Would a compromise be to parse the IEs as soon as the chip returns them, and then just store the parsed results. Obviously this is less flexible than returning everything the chip does but I'd imagine most people just want to distinguish between open networks, psk/personal networks, and enterprise networks (supporting the latter is a whole can of works of it's own of course),

plugwash, 2025-03-12:
I started doing some experimenting, to see if the values looked sane, and see how big a buffer would likely be needed if we were to save them. I got

Network eduroam 124 128 21757952

Where the three numbers are supposed to be the size of the BSSInfo struct and the ie_offset and ie_length fields. The first two look sane, but the latter does not.

I compared the C and rust structure definitions and noticed that the rust struct had #[repr(C, packed(2))] while the C struct did not. As a result it looks like we need to add two extra padding bytes before ie_length to get a sensible value.

plugwash, 2025-03-12:
diff --git a/cyw43/src/structs.rs b/cyw43/src/structs.rs
    index 81ae6a98d..4ab646f45 100644
    --- a/cyw43/src/structs.rs
    +++ b/cyw43/src/structs.rs
    @@ -534,6 +534,7 @@ pub struct BssInfo {
         pub basic_mcs: [u8; 16],
         /// Information Elements (IE) offset.
         pub ie_offset: u16,
    +    reserved16: u16,
         /// Length of Information Elements (IE) in bytes.
         pub ie_length: u32,
         /// Average signal-to-noise (SNR) ratio during frame reception.