embassy-rs/embassy #3167: embassy-boot can't compile with defmt feature

darkwater, 2024-07-11:
#3142 made some assertions const, but this doesn't work with defmt:

```
error[E0015]: cannot call non-const fn `defmt::export::acquire` in constants
   --> /home/dark/.cargo/git/checkouts/embassy-9312dcb0ed774b29/ed3da17/embassy-boot/src/fmt.rs:16:13
    |
16  |             ::defmt::assert!($($x)*);
    |             ^^^^^^^^^^^^^^^^^^^^^^^^
    |
   ::: /home/dark/.cargo/git/checkouts/embassy-9312dcb0ed774b29/ed3da17/embassy-boot/src/boot_loader.rs:239:13
    |
239 |             assert!(Self::PAGE_SIZE % ACTIVE::WRITE_SIZE as u32 == 0);
    |             --------------------------------------------------------- in this macro invocation
    |
    = note: calls in constants are limited to constant functions, tuple structs and tuple variants
    = note: this error originates in the macro `::defmt::assert` which comes from the expansion of the macro `assert` (in Nightly builds, run with -Z macro-backtrace for more info)
```

lulf, 2024-07-12:
Oh noes 😭 I think its ok to revert for now then, it was only a test.

lulf, 2024-08-02:
Fixed in https://github.com/embassy-rs/embassy/pull/3176

closed 2024-08-02 by lulf