embassy-rs/embassy #3178: PWM set_period issue

Gibbz, 2024-07-14:
Im trying to use PWM with a piezo to make a tune.
It plays an initial beep with the first set_duty ```pwm.set_duty(0, max_duty / 500);```. But after this it fails to play any more sound. 
The set_period or the enable doesnt work, im unsure which.
I'm using a nrf5840. 

```rust
use embassy_nrf::gpio::{AnyPin, OutputDrive};
use defmt::*;
use embassy_nrf::peripherals::PWM0;
use embassy_nrf::pwm::{Prescaler, SimplePwm};
use embassy_time::Timer;

const TASK_ID: &str = "PIEZO";

#[embassy_executor::task]
pub async fn piezo (
    pwm_device: PWM0,
    pin: AnyPin
) {
    info!("{}: start", TASK_ID);

    // note, frequency (hz)
    let tones = [
        ('c', 261),
        ('d', 294),
        ('e', 329),
        ('f', 349),
        ('g', 392),
        ('a', 440),
        ('b', 493),
    ];

    // note, duration
    let tune = [
        ('c', 1),
        ('c', 1),
        ('g', 1),
        ('g', 1),
        ('a', 1),
        ('a', 1),
        ('g', 2),
        ('f', 1),
        ('f', 1),
        ('e', 1),
        ('e', 1),
        ('d', 1),
        ('d', 1),
        ('c', 2),
        (' ', 4),
    ];

    let tempo = 300_u64; // ms
    let mut pwm = SimplePwm::new_1ch(pwm_device, pin);
    let max_duty = pwm.max_duty();
    pwm.set_duty(0, max_duty / 500); // 50% of max duty

    loop {
        info!("loop start");

        // 1. Obtain a note in the tune
        for note in tune {
            info!("note: {}", note.0);
            
            // 2. Retrieve the freqeuncy and beat associated with the note
            for tone in tones {
                // 2.1 Find a note match in the tones array and update frequency and beat variables accordingly
                if tone.0 == note.0 {
                    info!("tone: {}", tone.0);
                    // 3. Play the note for the desired duration (beats*tempo)
                    // 3.1 Adjust period of the PWM output to match the new frequency
                    pwm.set_period(tone.1);
 
                    // 3.2 Enable the channel to generate desired PWM
                    pwm.enable();

                    // 3.3 Keep the output on for as long as required
                    Timer::after_millis(note.1 * tempo).await;
                } else if note.0 == ' ' {
                    // 2.2 if ' ' tone is found disable output for one beat
                    pwm.disable();
                    Timer::after_millis(tempo).await;
                }
            }
            // 4. Silence for half a beat between notes
            // 4.1 Disable the PWM output (silence)
            pwm.disable();
            // 4.2 Keep the output off for half a beat between notes
            Timer::after_millis(tempo / 2).await;
            // 5. Go back to 1.
        }
    
    }

}
```

Dirbaio, 2024-07-14:
> `pwm.set_duty(0, max_duty / 500); // 50% of max duty`

to get 50% duty you want `max_duty/2`. Also, `max_duty` changes with frequency, so you should do this after every `.set_period()`.

> `pwm.set_period(tone.1);`

the period is not the frequency, it's the amount of 16mhz ticks in one cycle. to convert from frequency try `16_000_000 / tone.1`

Gibbz, 2024-07-14:
Thanks! Ive also had to set ```pwm.enabled()`` before the set_period and set_duty.
The period ```16_000_000 / tone.1``` didnt play any sound, but just using ```tone.1``` works.

Full code for anyone who stumbles here:

```rust
use embassy_nrf::gpio::{AnyPin, OutputDrive};
use defmt::*;
use embassy_nrf::peripherals::PWM0;
use embassy_nrf::pwm::{Prescaler, SimplePwm};
use embassy_time::Timer;

const TASK_ID: &str = "PIEZO";

#[embassy_executor::task]
pub async fn piezo (
    pwm_device: PWM0,
    pin: AnyPin
) {
    info!("{}: start", TASK_ID);

    // note, frequency (hz)
    let tones = [
        ('c', 261),
        ('d', 294),
        ('e', 329),
        ('f', 349),
        ('g', 392),
        ('a', 440),
        ('b', 493),
    ];

    // note, duration
    let tune = [
        ('c', 1),
        ('c', 1),
        ('g', 1),
        ('g', 1),
        ('a', 1),
        ('a', 1),
        ('g', 2),
        ('f', 1),
        ('f', 1),
        ('e', 1),
        ('e', 1),
        ('d', 1),
        ('d', 1),
        ('c', 2),
        (' ', 4),
    ];

    let tempo = 300_u64; // ms
    let mut pwm = SimplePwm::new_1ch(pwm_device, pin);
    let duty = pwm.max_duty() / 2; // piezo driver suggests 50% of max duty
    pwm.set_duty(0, duty);

    loop {
        info!("loop start");

        for note in tune {
            
            for tone in tones {
                if tone.0 == note.0 {
                    info!("tone: {}, {}", tone.0, tone.1);
                    // Play the note for the desired duration (beats*tempo)
                    // Adjust period of the PWM output to match the new frequency
                    // the period is not the frequency, it's the amount of 16mhz ticks in one cycle
                    pwm.enable();
                    pwm.set_period(tone.1); // TODO: try 16_000_000 / tone.1
                    pwm.set_duty(0, duty); // duty changes with period, so needs to be reset each time
                    Timer::after_millis(note.1 * tempo).await;
                } else if note.0 == ' ' {
                    // if ' ' tone is found disable output for one beat
                    pwm.disable();
                    Timer::after_millis(tempo).await;
                }
            }
            // Silence for half a beat between notes
            pwm.disable();
            Timer::after_millis(tempo / 2).await;
        }
    }
}
```

closed 2024-07-14 by Gibbz