embassy-rs/embassy #3180: Embassy-RP and RP2040-hal Compatibility??? - Embedded_Hal::PWM:SetDutyCycle Assistance

Token-Thinker, 2024-07-15:
Hello All,

I'm building a project based on `embedded-hal` and `embassy` for various boards that does async operations. 
When doing research I noticed that the `embassy-rp` crate doesn't expose the `channel` functionality for setting the duty cycle using `embedded-hal`. 

However `rp2040-hal` does expose that feature, but doesn't seem to have any ties to embassy, which leads to my issue/questions.

Is there a way for `embassy-rp` to exposes that or do I have to use `rp2040-hal` and if so will it work with embassy?

Any potential plans on merging the two?

Token-Thinker, 2024-07-24:
from the rp-hal side they said they don't see any plans on merging the two. 

thoughts from embassy maintainers? - https://github.com/rp-rs/rp-hal/issues/816

lulf, 2024-08-02:
These are different HALs/projects with different goals, so it's fine for them to coexist. If there is some functionality you'd like to have exposed (I'm not sure what `channel` functionality relates to, PWM?), please give some more details and/or raise a PR.

Token-Thinker, 2024-08-09:
@lulf 

that makes sense as i was reading more into it.

essentially, i want to be able to expose the `SetDutyCycle` function using the `embedded-hal` crate as a source of truth for my project that will be using multiple `mcus` 

the problem is in the implementation for `PWM` as it relates to `embassy-rp2040`, it directly slices and implements the "channel" control directly to the `io` vs letting you attach it like the `rp-hal` [example](https://github.com/rp-rs/rp-hal/blob/main/rp2040-hal/examples/pwm_blink.rs) below.

I hope that makes sense, thank you again for your response.

```rust
//! # PWM Blink Example
//.....omitting some setup configurations and imports 

#![no_std]
#![no_main]

use embedded_hal::pwm::SetDutyCycle;

fn main() -> ! {
    //.....omitting some setup configurations

    // Output channel B on PWM4 to GPIO 25
    let channel = &mut pwm.channel_b;
    channel.output_to(pins.gpio25);

    // Infinite loop, fading LED up and down
    loop {
        // Ramp brightness up
        for i in LOW..=HIGH {
            delay.delay_us(8);
            let _ = channel.set_duty_cycle(i); //<---------HERE
        }

        // Ramp brightness down
        for i in (LOW..=HIGH).rev() {
            delay.delay_us(8);
            let _ = channel.set_duty_cycle(i); //<----------HERE
        }

        delay.delay_ms(500);
    }
}

```

decaday, 2024-10-27:
https://github.com/embassy-rs/embassy/blob/b31648f2e59a68748e1c0f1408d1658a2e5e6a08/examples/rp/src/bin/pwm.rs#L13
https://github.com/embassy-rs/embassy/blob/b31648f2e59a68748e1c0f1408d1658a2e5e6a08/examples/rp/src/bin/pwm.rs#L67-L69

Token-Thinker, 2024-10-28:
> https://github.com/embassy-rs/embassy/blob/b31648f2e59a68748e1c0f1408d1658a2e5e6a08/examples/rp/src/bin/pwm.rs#L13
> 
> 
> https://github.com/embassy-rs/embassy/blob/b31648f2e59a68748e1c0f1408d1658a2e5e6a08/examples/rp/src/bin/pwm.rs#L67-L69

This could work! Thank you @decaday

closed 2024-10-28 by Token-Thinker