embassy-rs/embassy #3197: Feature Request: Cancel embassy executor task / `replace_spawn`

Ddystopia, 2024-07-18:
Hello,

In my use case I would like to have something like `embassy_executor::Spawner::replace_spawn`, that will always succeed. If it cannot spawn a new task, it would cancel some already existing one.

Gibbz, 2024-08-17:
On a similar note. I found it would be handy to have access to pausing, resuming, restarting and cancelling tasks by getting a handle. 

I have a project when I need to restart about 12 tasks and if I could cancel and recreate the tasks it would be much neater than having to have the same large blocks of code 12 times in each task.

zpg6, 2024-12-29:
> On a similar note. I found it would be handy to have access to pausing, resuming, restarting and cancelling tasks by getting a handle.
>
>
>
> I have a project when I need to restart about 12 tasks and [...]

I have a similar use case, but I would need some of my tasks to be able to finish the slice of work they are on and/or cleanup before terminating.