embassy-rs/embassy #3202: Way to guarantee stack space

trueb2, 2024-07-22:
Stack overflow is undefined behavior: https://github.com/embassy-rs/embassy/issues/3061#issuecomment-2165787740

Is there a way to guarantee that there will be roughly enough stack space? My guess is to do something to the linker script, but I am not very skilled with linker scripts.

I found myself trying to track down too much RAM usage, when I would have preferred failure to compile. From the help docs, I see that we can pursue `cargo size`, `cargo nm`, and `cargo-call-stack`. I would prefer if the application just didn't compile, if I don't have N KiB of stack space for my application.

Dirbaio, 2024-07-22:
you can modify the linker script to reserve space for the stack. Make the RAM memory region smaller, add another named STACK, then place stack start at the end of STACK instead of RAM. This will make linking fail if the RAM memory region overflows.

this is a cortex-m-rt thing though, not an embassy thing.

trueb2, 2024-07-22:
Thanks, @Dirbaio I am running with this now on the nRF52840 w/ softdevice

```
MEMORY
{
  /* NOTE 1 K = 1 KiBi = 1024 bytes */
  MBR                               : ORIGIN = 0x00000000, LENGTH = 4K
  SOFTDEVICE                        : ORIGIN = 0x00001000, LENGTH = 152K
  FLASH                             : ORIGIN = 0x00027000, LENGTH = 416K
  DFU                               : ORIGIN = 0x0008F000, LENGTH = 420K
  BOOTLOADER                        : ORIGIN = 0x000f9000, LENGTH = 24K
  BOOTLOADER_STATE                  : ORIGIN = 0x000ff000, LENGTH = 4K
  RAM                         (rwx) : ORIGIN = 0x20000000 + 19584, LENGTH = 240K - 19584
  STACK                       (rwx) : ORIGIN = 0x20000000 + 240K, LENGTH = 16K
}

__bootloader_state_start = ORIGIN(BOOTLOADER_STATE);
__bootloader_state_end = ORIGIN(BOOTLOADER_STATE) + LENGTH(BOOTLOADER_STATE);

__bootloader_dfu_start = ORIGIN(DFU);
__bootloader_dfu_end = ORIGIN(DFU) + LENGTH(DFU);
```

closed 2024-07-22 by trueb2

Dirbaio, 2024-07-22:
you also need `_stack_start = ORIGIN(STACK) + LENGTH(STACK)`, otherwise stack will still start from the end of the "RAM" region, and the "STACK" region will stay unused.