embassy-rs/embassy #3203: [embassy-stm32] allow splitting flash into banks not just regions

Gerharddc, 2024-07-23:
I am currently trying to setup a system with embassy-boot on an stm32f439zi and am using https://docs.embassy.dev/embassy-stm32/git/stm32f439zi/flash/struct.Flash.html#method.into_blocking_regions to split the flash into different regions for my different partitions. So far this appears to be working, but I have been forced into using a scheme that I believe is leaving a full 256KB of useable flash memory on the table.

This stm32 has the following flash layout:

[image]

Presumably the 3 different regions in each bank as shown on https://docs.embassy.dev/embassy-stm32/git/stm32f439zi/flash/struct.FlashLayout.html corresponds to regions with homogenous sector sizes.

Currently embassy-boot can only accept one flash device per partition (see https://docs.embassy.dev/embassy-boot/git/default/struct.BootLoaderConfig.html#method.from_linkerfile_blocking) so I have ended up using Bank1Region3 for my Active partition and Bank2Region3 for my DFU partition as per the following memory.x file:

```
MEMORY
{
  FLASH             : ORIGIN = 0x08000000, LENGTH = 64K
  BOOTLOADER_STATE  : ORIGIN = 0x08010000, LENGTH = 64K
  ACTIVE            : ORIGIN = 0x08020000, LENGTH = 768K
  DFU               : ORIGIN = 0x08120000, LENGTH = 896K
  RAM               : ORIGIN = 0x20000000, LENGTH =  192K
  RAM2              : ORIGIN = 0x10000000, LENGTH =  64K
  OTP               : ORIGIN = 0x1fff7800, LENGTH =  528
}

/* All 3 sections have been placed at the beginning of their respective regions so have 0 offsets */

__bootloader_state_start = 0;
__bootloader_state_end = LENGTH(BOOTLOADER_STATE);

__bootloader_active_start = 0;
__bootloader_active_end = LENGTH(ACTIVE);

__bootloader_dfu_start = 0;
__bootloader_dfu_end = LENGTH(DFU);
```
The problem here is that the last 128KB of Bank1Region3 goes unused since ACTIVE needs to be 1 page smaller than DFU. Furthermore, Bank2Region1+Bank2Region2 (so the first 128KB of Bank2)  also goes unused because I have no way of presenting the entire Bank2 to embassy-boot as a single device where it can put the DFU partition. If that were possible, then I would be able to use both of those regions effectively to increase my max firmware size by 128KB.

Of course, an alternative is to not use `into_blocking_regions` and to instead present the entire flash device directly yo embassy-boot, but the problem then is that I end up with an ERASE_SIZE of 128KB which would force all partitions (including FLASH and BOOTLOADER_STATE) to be multiples of 128KB which would be quite wasteful and not help to increase ACTIVE in this particular situation.

A very nice solution in my opinion would be to have something like `into_blocking_regions` available that splits the flash device into banks instead of regions. Of course, it should then be possible to split an individual bank (bank 1 in my case) into regions afterwards.

Dirbaio, 2024-07-23:
you can glue multiple regions into one with [ConcatFlash](https://docs.embassy.dev/embassy-embedded-hal/git/default/flash/struct.ConcatFlash.html), then give the result to embassy-boot.

Gerharddc, 2024-07-23:
@Dirbaio great, thanks! That was the other possible solution that I could think of it but sounded more complex to implement. Great to see it already exists :)

closed 2024-07-23 by Gerharddc