embassy-rs/embassy #3204: Weird issue with `Timer::after_secs(99999999999999999).await;`

busslina, 2024-07-23:
I'm awaiting on the main task but a secondary task is getting blocked (or some error happened) with:
```
Timer::after_secs(99999999999999999).await;
```

With a smaller number it works okay:
```
Timer::after_secs(u32::MAX as u64).await;
```

Firstly discussed [here](https://matrix.to/#/!YoLPkieCYHGzdjUhOK:matrix.org/$PLq02hfOMKbJ_b3VV4n3zFyMlrz2NjIhPsnTy9uHZCg?via=matrix.org&via=tchncs.de&via=mozilla.org)

jamesmunns, 2024-07-24:
For reference:

```
>>> hex(99999999999999999)
'0x16345785d89ffff'
```

Which is:

`0x0163_4578_5d89_ffff`

and as `Duration` has a max tick range of `u64`:

```
>>> (2 ** 64) / 99999999999999999
184.46744073709553
```

So converting from seconds to ticks, we are almost certainly overflowing the tick count (embassy-rp defaults to 1MHz, iirc):

```
>>> hex((99999999999999999 * 1000000) & ((2 ** 64) - 1))
'0x2c7_e14a_f670_bdc0'
```

I'm not sure what the cause of the lock up is though.

busslina, 2024-07-24:
I tried to get an error message with this, but nothing happens, it just hangs:

```rs
#[panic_handler]
fn on_panic(info: &PanicInfo) -> ! {
    log::error!("Panic: {}", info);
    loop {}
}
```

Dirbaio, 2024-07-24:
> Debug: Usb

the USB logger won't log panic messages, because panics crash the firmware and usb stops working if the firmware stops running. To get panic messages you need to use a debug probe.

Dirbaio, 2024-07-24:
Panic message is

```
ERROR - panicked at /home/dirbaio/embassy/embassy/embassy-time/src/duration.rs:47:27:
attempt to multiply with overflow
```

This is working as intended. Duration is a u64 number of ticks, conversion from seconds can overflow.

Note if you want to sleep forever you can do `core::future::pending::<()>().await`.
Also note it's fine to let `main` return, other tasks continue running.

closed 2024-07-24 by Dirbaio

busslina, 2024-07-24:
@Dirbaio Thanks