embassy-rs/embassy #3205: STM32H7 recover from SAI `OverrunError`

elagil, 2024-07-24:
Hello!

I use the `embassy_stm32::sai::SAI4` peripheral driver on the `STM32H723` for writing audio data to a TDM output.

I have to start writing data immediately after calling `start()` on the driver instance - then it works fine. Otherwise, the writable ring buffer gives an `OverrunError`, which I cannot recover from, because subsequent calls to `write()` also give that error immediately. The same happens when there is a pause in writing, and (I guess) that the DMA consumes all of the data in the ring buffer.

I did not find a way to reset the ring buffer (and probably the driver itself). Do I have to make a new driver instance each time? Otherwise, I can also send zeros to the instance, to keep the buffer entertained, but I would rather be able to switch the peripheral off sometimes.

Thanks!

lulf, 2024-08-02:
I'm not familiar with SAI, but generally all drivers can be passed a &mut instance upon creation, and when the driver goes out of scope, it switches off the peripheral. Then you can reuse the peripheral with the driver next time you want to use it.

elagil, 2024-08-02:
Thanks, I tried that and it works fine. I think I will stick with it.

closed 2024-08-02 by elagil

I--P, 2024-10-22:
Could you please elaborate on how you fixed that ? I'm running into the same issue with read() and I don't understand how to get rid of the overrun error

elagil, 2024-10-23:
@I--P You can have a look at this implementation that I ended up with: https://github.com/blus-audio/firmware-rs/blob/main/blus_mini_mk2/src/audio_routing.rs#L325

When there is an error, a fresh SAI instance is created with `new_sai4()`.

I--P, 2024-10-23:
@elagil thanks a lot, that's what I ended up with too. It will be very helpful for many people to have a direct link to your code here.