embassy-rs/embassy #3208: gpiote_port example fails

Syphixs, 2024-07-25:
Example  [examples/nrf52840/src/bin/gpiote_port.rs] fails and cannot add generic parameter easily. AnyPin does not work either.

```rust
#![no_std]
#![no_main]

use defmt::{info, unwrap};
use embassy_executor::Spawner;
use embassy_nrf::gpio::{Input, Pull};
use {defmt_rtt as _, panic_probe as _};

#[embassy_executor::task(pool_size = 4)]
async fn button_task(n: usize, mut pin: Input<'static>) {
    loop {
        pin.wait_for_low().await;
        info!("Button {:?} pressed!", n);
        pin.wait_for_high().await;
        info!("Button {:?} released!", n);
    }
}

#[embassy_executor::main]
async fn main(spawner: Spawner) {
    let p = embassy_nrf::init(Default::default());
    info!("Starting!");

    let btn1 = Input::new(p.P0_11, Pull::Up);
    let btn2 = Input::new(p.P0_12, Pull::Up);
    let btn3 = Input::new(p.P0_24, Pull::Up);
    let btn4 = Input::new(p.P0_25, Pull::Up);

    unwrap!(spawner.spawn(button_task(1, btn1)));
    unwrap!(spawner.spawn(button_task(2, btn2)));
    unwrap!(spawner.spawn(button_task(3, btn3)));
    unwrap!(spawner.spawn(button_task(4, btn4)));
}

```
Error:
```
struct takes 1 generic argument but 0 generic arguments were supplied
   expected 1 generic argument [E0107]
```

i saw the last commit just removed the generics in it. is this not possible anymore?

Dirbaio, 2024-07-25:
you're using an example from latest git, with embassy-nrf 0.1 from crates.io. Examples from latest git work with embassy-nrf from latest git.

Syphixs, 2024-07-26:
You are completely right. Thanks and sorry for my oversight.

closed 2024-07-26 by Syphixs