embassy-rs/embassy #3220: Cellular/4G network?

overflowz, 2024-07-29:
I'm fairly new to embedded land, please correct me if I'm asking some nonsense.

I'm learning how everything works around and I faced this issue: I have STM32 chip and A7670 4G module. What I'm trying to achieve is, to connect to the internet, keep TCP connection open and send/receive packets over it -- with MQTT client).

My problem is, I have no idea how to actually make it work. I also tried searching over the internet, but with no luck (to use it with either Rust or Embassy).

So the main question, does embassy-net support cellular network, if so, where or what exactly should I look for?

Any tips would be much appreciated.

Thank you.

lulf, 2024-08-02:
Hi! Embassy-net combines a TCP/IP stack (smotcp) with an async driver for the underlying networking interface. 

There is no embassy-net driver for your particular module, so that would need to be implemented in order for it to be used with embassy-net. You may have success using embassy-ppp to build a driver for your module (I'm not familiar with A7670 so not sure if it supports serial PPP or something like that). You can also look at the embassy-net-* drivers in the repo for inspiration.

Feel free to reach out in the matrix.org chat if you want to discuss this further.

overflowz, 2024-08-02:
Thanks for the reply!

I'm guessing this is kinda related to this one here: https://github.com/embassy-rs/embassy/issues/3096

By doing more and more research, if I'm able to switch to PPP (I had no idea what this was), it should work fine. I'll test it whenever I can to make sure it works and can also provide an example so it won't be confusing for newcomers too.

Thank you so much!

closed 2024-08-02 by overflowz