embassy-rs/embassy #3236: rp: wifi_tcp_server: fails with `dns` feature of `embassy-net`

marxin, 2024-08-07:
Technically speaking it started with b05217b3566e17cd10eb52c714aa53fc6d52dd91, where the `dns` feature was added to `examples/rp/Cargo.toml`:

<details>
  <summary>Complete output</summary>

```
❯ cargo run  --bin wifi_tcp_server
   Compiling embassy-rp-examples v0.1.0 (/home/marxin/Programming/embassy/examples/rp)
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.29s
     Running `probe-rs run --chip RP2040 target/thumbv6m-none-eabi/debug/wifi_tcp_server`
     Erasing sectors ✔ [00:00:07] [#########################################################################################################################################################################################################################################################################################] 632.00 KiB/632.00 KiB @ 84.42 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:21] [#########################################################################################################################################################################################################################################################################################] 632.00 KiB/632.00 KiB @ 29.53 KiB/s (eta 0s )    Finished in 28.927s
0.005373 INFO  Hello World!
└─ wifi_tcp_server::____embassy_main_task::{async_fn#0} @ src/bin/wifi_tcp_server.rs:43  
0.289829 DEBUG waiting for clock...
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
0.290778 DEBUG clock ok
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
0.293244 DEBUG chip ID: 43439
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
0.304740 DEBUG loading fw
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.224609 DEBUG loading nvram
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.232727 DEBUG starting up core...
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.259777 DEBUG waiting for wifi init...
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.346532 DEBUG shared_addr 000177ac
└─ cyw43::runner::{impl#1}::log_init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.348901 DEBUG wifi init done
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.350590 DEBUG Downloading CLM...
└─ cyw43::control::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.366440 DEBUG LOGS: 000000.001 
└─ cyw43::runner::{impl#1}::log_read::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.366907 DEBUG LOGS: RTE (SDIO-CDC) 7.95.62 (b03806e CY) on BCM43439 r5 @ 37.4/81.6/81.6MHz
└─ cyw43::runner::{impl#1}::log_read::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.368050 DEBUG LOGS: 000000.003 sdpcmdcdc0: Broadcom SDPCMD CDC driver
└─ cyw43::runner::{impl#1}::log_read::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.373529 DEBUG LOGS: 000000.055 wl0: Broadcom BCM43439 802.11 Wireless Controller 7.95.62 (b03806e CY)
└─ cyw43::runner::{impl#1}::log_read::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.374791 DEBUG LOGS: 000000.057 TCAM: 256 used: 77 exceed:0
└─ cyw43::runner::{impl#1}::log_read::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.380294 WARN  BDC event, incomplete data
└─ cyw43::runner::{impl#1}::rx @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:170 
2.382148 WARN  BDC event, incomplete data
└─ cyw43::runner::{impl#1}::rx @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:170 
2.386360 WARN  IRQ DATA_UNAVAILABLE, clearing...
└─ cyw43::runner::{impl#1}::handle_irq::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:170 
2.429494 DEBUG get clmload_status
└─ cyw43::control::{impl#1}::get_iovar::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.439654 DEBUG Configuring misc stuff...
└─ cyw43::control::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.439965 DEBUG set bus:txglom = [00, 00, 00, 00]
└─ cyw43::control::{impl#1}::set_iovar_v::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.450296 DEBUG set apsta = [01, 00, 00, 00]
└─ cyw43::control::{impl#1}::set_iovar_v::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.460498 DEBUG get cur_etheraddr
└─ cyw43::control::{impl#1}::get_iovar::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.470477 DEBUG mac addr: [d8, 3a, dd, 6c, 94, 39]
└─ cyw43::control::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.471100 DEBUG set country = [58, 58, 00, 00, ff, ff, ff, ff, 58, 58, 00, 00]
└─ cyw43::control::{impl#1}::set_iovar_v::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.611412 DEBUG set bus:txglom = [00, 00, 00, 00]
└─ cyw43::control::{impl#1}::set_iovar_v::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.722108 DEBUG set ampdu_ba_wsize = [08, 00, 00, 00]
└─ cyw43::control::{impl#1}::set_iovar_v::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
2.833160 DEBUG set ampdu_mpdu = [04, 00, 00, 00]
└─ cyw43::control::{impl#1}::set_iovar_v::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
[... 219 lines cut ...]
└─ embassy_net::{impl#5}::poll @ /home/marxin/Programming/embassy/embassy-net/src/fmt.rs:156 
6.721772 INFO  IPv4: DOWN
└─ embassy_net::{impl#5}::apply_static_config @ /home/marxin/Programming/embassy/embassy-net/src/fmt.rs:156 
6.723456 DEBUG DHCP send DISCOVER to 255.255.255.255: Repr { message_type: Discover, transaction_id: 3210669, secs: 0, client_hardware_address: Address([216, 58, 221, 108, 148, 57]), client_ip: 0.0.0.0, your_ip: 0.0.0.0, server_ip: 0.0.0.0, router: None, subnet_mask: None, relay_agent_ip: 0.0.0.0, broadcast: false, requested_ip: None, client_identifier: Some(Address([216, 58, 221, 108, 148, 57])), server_identifier: None, parameter_request_list: Some([1, 3, 6]), dns_servers: None, max_size: Some(1432), lease_duration: None, renew_duration: None, rebind_duration: None, additional_options: [] }
└─ smoltcp::socket::dhcpv4::{impl#1}::dispatch @ /home/marxin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
6.770097 DEBUG DHCP recv Offer from 192.168.0.1: Repr { message_type: Offer, transaction_id: 3210669, secs: 0, client_hardware_address: Address([216, 58, 221, 108, 148, 57]), client_ip: 0.0.0.0, your_ip: 192.168.0.4, server_ip: 192.168.0.1, router: Some(192.168.0.1), subnet_mask: Some(255.255.255.0), relay_agent_ip: 0.0.0.0, broadcast: false, requested_ip: None, client_identifier: None, server_identifier: Some(192.168.0.1), parameter_request_list: None, dns_servers: Some([195.39.77.66, 8.8.8.8]), max_size: None, lease_duration: Some(1800), renew_duration: None, rebind_duration: None, additional_options: [] }
└─ smoltcp::socket::dhcpv4::{impl#1}::process @ /home/marxin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
6.773498 DEBUG DHCP send request to 255.255.255.255: Repr { message_type: Request, transaction_id: 1316354120, secs: 0, client_hardware_address: Address([216, 58, 221, 108, 148, 57]), client_ip: 0.0.0.0, your_ip: 0.0.0.0, server_ip: 0.0.0.0, router: None, subnet_mask: None, relay_agent_ip: 0.0.0.0, broadcast: false, requested_ip: Some(192.168.0.4), client_identifier: Some(Address([216, 58, 221, 108, 148, 57])), server_identifier: Some(192.168.0.1), parameter_request_list: Some([1, 3, 6]), dns_servers: None, max_size: Some(1432), lease_duration: None, renew_duration: None, rebind_duration: None, additional_options: [] }
└─ smoltcp::socket::dhcpv4::{impl#1}::dispatch @ /home/marxin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
6.798008 DEBUG DHCP recv Ack from 192.168.0.1: Repr { message_type: Ack, transaction_id: 1316354120, secs: 0, client_hardware_address: Address([216, 58, 221, 108, 148, 57]), client_ip: 0.0.0.0, your_ip: 192.168.0.4, server_ip: 192.168.0.1, router: Some(192.168.0.1), subnet_mask: Some(255.255.255.0), relay_agent_ip: 0.0.0.0, broadcast: false, r [... cut ...]
└─ embassy_net::{impl#5}::apply_static_config @ /home/marxin/Programming/embassy/embassy-net/src/fmt.rs:142 
6.820613 INFO  DHCP is now up!
└─ wifi_tcp_server::____embassy_main_task::{async_fn#0} @ src/bin/wifi_tcp_server.rs:109 
6.821750 DEBUG set gpioout = [01, 00, 00, 00, 00, 00, 00, 00]
└─ cyw43::control::{impl#1}::set_iovar_v::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:142 
6.833515 INFO  Listening on TCP:1234...
└─ wifi_tcp_server::____embassy_main_task::{async_fn#0} @ src/bin/wifi_tcp_server.rs:122 
```
</details>

Dirbaio, 2024-08-07:
are you using latest git version? I think this was fixed in https://github.com/embassy-rs/embassy/pull/3219 .

Also note https://github.com/embassy-rs/embassy/pull/3232 fixes the examples to have enough socket slots.

marxin, 2024-08-07:
Thanks! I can confirm it's fixed now with the current main branch.

closed 2024-08-07 by marxin