embassy-rs/embassy #3237: rp: wifi_tcp_server: fails after the recent `bluetooth` feature for `cyw43`

marxin, 2024-08-07:
After 891c5ee10584cd990dad529e3506fe1328e4e69d, `wifi_tcp_server` stucks here:

```
0.002149 INFO  Hello World!
└─ wifi_tcp_server::____embassy_main_task::{async_fn#0} @ src/bin/wifi_tcp_server.rs:45  
0.272004 DEBUG init alp
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.272054 DEBUG set f2 watermark
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.272125 DEBUG watermark = 10
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.272157 DEBUG waiting for clock...
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.272207 DEBUG clock ok
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.272228 DEBUG clear request for ALP
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.272402 DEBUG chip ID: 43439
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.275007 DEBUG loading fw
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.455116 DEBUG loading nvram
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.455869 DEBUG starting up core...
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.458456 DEBUG wait for HT clock
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.476906 DEBUG setup interrupt mask
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.477032 DEBUG waiting for F2 to be ready...
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.562428 DEBUG clear pad pulls
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.562526 DEBUG waiting for HT clock...
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.562576 DEBUG clock ok
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.562684 DEBUG shared_addr 00017a7c
└─ cyw43::runner::{impl#1}::log_init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.562908 DEBUG cyw43 runner init done
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.563112 DEBUG Downloading CLM...
└─ cyw43::control::{impl#1}::load_clm::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.564519 DEBUG LOGS: hndarm_armr addr: 0x18003000, cr4_idx: 0
└─ cyw43::runner::{impl#1}::log_read::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.564671 DEBUG LOGS: 000000.001 
└─ cyw43::runner::{impl#1}::log_read::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.564778 DEBUG LOGS: RTE (SDIO-CDC) 7.95.60 (abcd531 CY) on BCM43439 r5 @ 37.4/81.6/81.6MHz
└─ cyw43::runner::{impl#1}::log_read::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.564998 DEBUG LOGS: 000000.003 sdpcmdcdc0: Broadcom SDPCMD CDC driver
└─ cyw43::runner::{impl#1}::log_read::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.565181 DEBUG LOGS: 000000.008 reclaim section 0: Returned 46156 bytes to the heap
└─ cyw43::runner::{impl#1}::log_read::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.565378 DEBUG LOGS: 000000.012 wlc_bmac_info_init: host_enab 1
└─ cyw43::runner::{impl#1}::log_read::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.565557 DEBUG LOGS: 000000.039 wl0: wlc_ampdu_tx_set: AGG Mode = MAC+Ucode txmaxpkts 64 txmaxpkts_agg 0
└─ cyw43::runner::{impl#1}::log_read::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.565828 DEBUG LOGS: 000000.050 wl0: wlc_channels_commit: no valid channel for "#n" nbands 1 bandlocked 0
└─ cyw43::runner::{impl#1}::log_read::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.566099 DEBUG LOGS: 000000.055 wl0: Broadcom BCM43439 802.11 Wireless Controller 7.95.61 (abcd531 CY)
└─ cyw43::runner::{impl#1}::log_read::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.566336 DEBUG LOGS: 000000.057 TCAM: 256 used: 77 exceed:0
└─ cyw43::runner::{impl#1}::log_read::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.566493 DEBUG LOGS: 000000.059 reclaim section 1: Returned 86132 bytes to the heap
└─ cyw43::runner::{impl#1}::log_read::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
0.567279 WARN  BDC event, incomplete header
└─ cyw43::runner::{impl#1}::rx @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:166 
0.567397 WARN  BDC event, incomplete header
└─ cyw43::runner::{impl#1}::rx @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:166 
```
[... 168 lines cut ...]
14.344828 DEBUG DHCP recv Offer from 192.168.0.1: Repr { message_type: Offer, transaction_id: 346788866, secs: 0, client_hardware_address: Address([216, 58, 221, 108, 148, 57]), client_ip: 0.0.0.0, your_ip: 192.168.0.4, server_ip: 192.168.0.1, router: Some(192.168.0.1), subnet_mask: Some(255.255.255.0), relay_agent_ip: 0.0.0.0, broadcast: false, requested_ip: None, client_identifier: None, server_identifier: Some(192.168.0.1), parameter_request_list: None, dns_servers: Some([195.39.77.66, 8.8.8.8]), max_size: None, lease_duration: Some(1800), renew_duration: None, rebind_duration: None, additional_options: [] }
└─ smoltcp::socket::dhcpv4::{impl#1}::process @ /home/marxin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
14.345359 DEBUG DHCP send request to 255.255.255.255: Repr { message_type: Request, transaction_id: 889153170, secs: 0, client_hardware_address: Address([216, 58, 221, 108, 148, 57]), client_ip: 0.0.0.0, your_ip: 0.0.0.0, server_ip: 0.0.0.0, router: None, subnet_mask: None, relay_agent_ip: 0.0.0.0, broadcast: false, requested_ip: Some(192.168.0.4), client_identifier: Some(Address([216, 58, 221, 108, 148, 57])), server_identifier: Some(192.168.0.1), parameter_request_list: Some([1, 3, 6]), dns_servers: None, max_size: Some(1432), lease_duration: None, renew_duration: None, rebind_duration: None, additional_options: [] }
└─ smoltcp::socket::dhcpv4::{impl#1}::dispatch @ /home/marxin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
14.352009 DEBUG DHCP recv Ack from 192.168.0.1: Repr { message_type: Ack, transaction_id: 889153170, secs: 0, client_hardware_address: Address([216, 58, 221, 108, 148, 57]), client_ip: 0.0.0.0, your_ip: 192.168.0.4, server_ip: 192.168.0.1, router: Some(192.168.0.1), subnet_mask: Some(255.255.255.0), relay_agent_ip: 0.0.0.0, broadcast: false, requested_ip: None, client_identifier: None, server_identifier: Some(192.168.0.1), parameter_request_list: None, dns_servers: Some([195.39.77.66, 8.8.8.8]), max_size: None, lease_duration: Some(1800), renew_duration: None, rebind_duration: None, additional_options: [] }
└─ smoltcp::socket::dhcpv4::{impl#1}::process @ /home/marxin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/macros.rs:18  
14.352702 DEBUG IPv4: UP
└─ embassy_net::{impl#5}::apply_static_config @ /home/marxin/Programming/embassy/embassy-net/src/fmt.rs:138 
14.352729 DEBUG    IP address:      192.168.0.4/24
└─ embassy_net::{impl#5}::apply_static_config @ /home/marxin/Programming/embassy/embassy-net/src/fmt.rs:138 
14.352787 DEBUG    Default gateway: Some(192.168.0.1)
└─ embassy_net::{impl#5}::apply_static_config @ /home/marxin/Programming/embassy/embassy-net/src/fmt.rs:138 
14.352847 DEBUG    DNS server:      195.39.77.66
└─ embassy_net::{impl#5}::apply_static_config @ /home/marxin/Programming/embassy/embassy-net/src/fmt.rs:138 
14.352894 DEBUG    DNS server:      8.8.8.8
└─ embassy_net::{impl#5}::apply_static_config @ /home/marxin/Programming/embassy/embassy-net/src/fmt.rs:138 
14.352991 WARN  Number of DNS servers exceeds DNS_MAX_SERVER_COUNT, truncating list.
└─ embassy_net::{impl#5}::apply_static_config @ /home/marxin/Programming/embassy/embassy-net/src/fmt.rs:166 
14.418639 INFO  DHCP is now up!
└─ wifi_tcp_server::____embassy_main_task::{async_fn#0} @ src/bin/wifi_tcp_server.rs:112 
14.418884 DEBUG set gpioout = [01, 00, 00, 00, 00, 00, 00, 00]
└─ cyw43::control::{impl#1}::set_iovar_v::{async_fn#0} @ /home/marxin/Programming/embassy/cyw43/src/fmt.rs:138 
14.420027 INFO  Listening on TCP:1234...
└─ wifi_tcp_server::____embassy_main_task::{async_fn#0} @ src/bin/wifi_tcp_server.rs:125 
```

Dirbaio, 2024-08-07:
the wifi examples are working for me on latest git main. Are you running the examples unmodified from a fresh git clone, including latest cyw43 firmware?

marxin, 2024-08-08:
I'm running on the main branch (891c5ee10584cd990dad529e3506fe1328e4e69d) with the only change related to my WiFi password:

working version:

```patch
diff --git a/examples/rp/Cargo.toml b/examples/rp/Cargo.toml
index 2884ca85a..acfb1fd84 100644
--- a/examples/rp/Cargo.toml
+++ b/examples/rp/Cargo.toml
@@ -16,7 +16,7 @@ embassy-net = { version = "0.4.0", path = "../../embassy-net", features = ["defm
 embassy-net-wiznet = { version = "0.1.0", path = "../../embassy-net-wiznet", features = ["defmt"] }
 embassy-futures = { version = "0.1.0", path = "../../embassy-futures" }
 embassy-usb-logger = { version = "0.2.0", path = "../../embassy-usb-logger" }
-cyw43 = { version = "0.2.0", path = "../../cyw43", features = ["defmt", "firmware-logs", "bluetooth"] }
+cyw43 = { version = "0.2.0", path = "../../cyw43", features = ["defmt", "firmware-logs"] }
 cyw43-pio = { version = "0.2.0", path = "../../cyw43-pio", features = ["defmt"] }
 
 defmt = "0.3"
diff --git a/examples/rp/src/bin/wifi_tcp_server.rs b/examples/rp/src/bin/wifi_tcp_server.rs
index 61eeb82f7..d7ef7af42 100644
--- a/examples/rp/src/bin/wifi_tcp_server.rs
+++ b/examples/rp/src/bin/wifi_tcp_server.rs
@@ -27,8 +27,8 @@ bind_interrupts!(struct Irqs {
     PIO0_IRQ_0 => InterruptHandler<PIO0>;
 });
 
-const WIFI_NETWORK: &str = "LadronDeWifi";
-const WIFI_PASSWORD: &str = "MBfcaedHmyRFE4kaQ1O5SsY8";
+const WIFI_NETWORK: &str = "marxin";
+const WIFI_PASSWORD: &str = "xxxxxxxxxxx";
 
 #[embassy_executor::task]
 async fn cyw43_task(runner: cyw43::Runner<'static, Output<'static>, PioSpi<'static, PIO0, 0, DMA_CH0>>) -> ! {
```

not working version:

```patch
diff --git a/examples/rp/src/bin/wifi_tcp_server.rs b/examples/rp/src/bin/wifi_tcp_server.rs
index 61eeb82f7..d7ef7af42 100644
--- a/examples/rp/src/bin/wifi_tcp_server.rs
+++ b/examples/rp/src/bin/wifi_tcp_server.rs
@@ -27,8 +27,8 @@ bind_interrupts!(struct Irqs {
     PIO0_IRQ_0 => InterruptHandler<PIO0>;
 });
 
-const WIFI_NETWORK: &str = "LadronDeWifi";
-const WIFI_PASSWORD: &str = "MBfcaedHmyRFE4kaQ1O5SsY8";
+const WIFI_NETWORK: &str = "marxin";
+const WIFI_PASSWORD: &str = "xxxxxxxxxxx";
 
 #[embassy_executor::task]
 async fn cyw43_task(runner: cyw43::Runner<'static, Output<'static>, PioSpi<'static, PIO0, 0, DMA_CH0>>) -> ! {
```

I'm using the following compiler:
```
❯ rustc --version
rustc 1.80.0 (051478957 2024-07-21)
```

and I'm attaching the output with `DEFMT_LOG=trace`, it might help:
[good.txt](https://github.com/user-attachments/files/16540418/good.txt)
[bad.txt](https://github.com/user-attachments/files/16540420/bad.txt)

Dirbaio, 2024-08-08:
Interesting, it works with --release but not without. Seems it's broken depending on the optimization flags:

lto=false opt-level=3 -> works
lto=false opt-level=s -> fails
lto=false opt-level=z -> fails
lto=true opt-level=3 -> works
lto=true opt-level=s -> works
lto=true opt-level=z -> fails

i'll take a look. As a workaround for now, either disable bluetooth or use a set of flags that works.

marxin, 2024-08-08:
Yeah, sorry that I forgot to mention that I was testing the no-release mode ;) I can confirm `--release` mode helps here.
What's interesting is that the debug mode (-z + LTO) binary size is smaller than when you get with the release mode.

The binary limitation is mentioned in #3115 which is the PR that introduced LTO and -z for debug mode.

closed 2024-08-08 by Dirbaio

marxin, 2024-08-09:
Thanks! I can confirm it works now for me.

PR #3242 (2024-08-08): cyw43: make sure to yield if doing busy-polling for interrupts.
fixes #3237