embassy-rs/embassy #3241: RP2350 support

brandonros, 2024-08-08:
Do we know enough yet what we need to do to be able to support [RP2350](https://www.raspberrypi.com/news/raspberry-pi-pico-2-our-new-5-microcontroller-board-on-sale-now/) instead of just RP2040?

CBJamo, 2024-08-09:
Initial support is coming in #3243, though much work remains.

BjornTheProgrammer, 2024-08-12:
I've preordered the raspberry pi pico 2 and would be willing to test it on actual hardware and write/verify some examples.

ionspin, 2024-08-20:
I've got a couple of Tiny2350 from pimoroni, got blinky to work without much trouble except that elf2uf2 would work because of hard float in the executable.

```
elf2uf2-rs ./target/thumbv8m.main-none-eabihf/debug/blinky                                                                                                                                                                            
Error: "HARD-FLOAT not supported"
```
Worked around that by using `picotool` and this small script:
`picotool-run.sh`
```bash
picotool uf2 convert -t elf $1 out.uf2
picotool load out.uf2
picotool reboot
```
in `.cargo/config.toml` for the `runner` 

This is a quick workaround until we get support in elf2uf2 for `thumbv8m.main-none-eabihf` or full probe-rs support.

CBJamo, 2024-08-20:
JP has a pull request for elf2uf2-rs to allow use with the rp2350. https://github.com/JoNil/elf2uf2-rs/pull/33

thejpster, 2024-08-22:
However, I recommend switching to `picotool` because it's faster and doesn't require mounting the disk drive. Binary releases are available from https://github.com/raspberrypi/pico-sdk-tools/releases

ionspin, 2024-08-23:
I'm running into a bit of issue, but I'm getting different behavior after `picotool reboot` and resseting the RP2350 (using reset button on Tiny2350, or just unplugging the Pico2). After `picotool reboot` it works fine every time, but the reset get's it into a _partially_ broken state.

This is more or less the current setup, it's reproducible if I remove the USB part and just leave the blinky loop
```rust
#[embassy_executor::main]
async fn main(spawner: Spawner) {
    let peripherals = embassy_rp::init(Default::default());

    let other_side_channel = OTHER_SIDE_CHANNEL.init(Channel::new());

    UsbHandler::new(peripherals.USB, other_side_channel, spawner, true);

    
    let delay = 100;

    let mut counter = 0;
    #[cfg(feature = "tiny2350")]
    {
        let mut red_led = Output::new(peripherals.PIN_18, Level::Low);
        let mut green_led = Output::new(peripherals.PIN_19, Level::Low);
        let mut blue_led = Output::new(peripherals.PIN_20, Level::Low);
        green_led.set_high();
        blue_led.set_high();
        red_led.set_high();
        loop {
            log::info!("Tick {}", counter);
            counter += 1;
            red_led.set_low();
            Timer::after_millis(delay).await;

            red_led.set_high();
            Timer::after_millis(delay).await;

            green_led.set_low();
            Timer::after_millis(delay).await;

            green_led.set_high();
            Timer::after_millis(delay).await;

            blue_led.set_low();
            Timer::after_millis(delay).await;

            blue_led.set_high();
            Timer::after_millis(delay).await;
        }
    }

    #[cfg(feature = "pico2")]
    {
        let mut led =  Output::new(peripherals.PIN_25, Level::Low);
        led.set_low();
        loop {
            log::info!("Tick {}", counter);
            counter += 1;
            led.set_high();
            Timer::after_millis(delay).await;
            led.set_low();
            Timer::after_millis(delay).await;
        }
    }

}
```
On `picotool reboot` the blinks and log-over-usb works fine, as expected, but when using reset on Tiny2350 or unplugging and plugging in on Pico2, the blinks are not there, while log-over-usb gets set up most of the time.

Any clues on why this is happening? I'm planning to solder the headers tomorrow so I can try to set up UART and see if I get more info that way.

ionspin, 2024-08-24:
I don't think this is correct:
https://github.com/embassy-rs/embassy/blob/423e5d7655d038ead9ad7f5f448cabbd0b330660/embassy-rp/src/clocks.rs#L515

[image]

I suspect `picotool reboot` initializes ticks, and embassy-rp does not, here's my log collected through defmt-uart

```

0.037927 INFO  Defmt serial test
0.038167 INFO  USB: config_descriptor used: 94
0.038175 INFO  USB: bos_descriptor used: 12
0.038182 INFO  USB: msos_descriptor used: 0
0.038188 INFO  USB: control_buf size: 256
0.038432 INFO  Defmt tick 0
0.238699 INFO  Defmt tick 1
0.364565 DEBUG SET_CONFIGURATION: configured
0.365786 DEBUG Set line coding to: LineCoding { stop_bits: One, data_bits: 8, parity_type: None, data_rate: 9600 }
0.438751 INFO  Defmt tick 2
0.537203 DEBUG Set dtr true, rts true
0.638785 INFO  Defmt tick 3
0.838805 INFO  Defmt tick 4
1.038821 INFO  Defmt tick 5
1.238837 INFO  Defmt tick 6
1.438854 INFO  Defmt tick 7
1.638870 INFO  Defmt tick 8
1.838886 INFO  Defmt tick 9
2.038903 INFO  Defmt tick 10
2.238919 INFO  Defmt tick 11

0.000000 INFO  Defmt serial test
0.000000 INFO  USB: config_descriptor used: 94
0.000000 INFO  USB: bos_descriptor used: 12
0.000000 INFO  USB: msos_descriptor used: 0
0.000000 INFO  USB: control_buf size: 256
0.000000 INFO  Defmt tick 0
0.000000 DEBUG SET_CONFIGURATION: configured
0.000000 DEBUG Set line coding to: LineCoding { stop_bits: One, data_bits: 8, parity_type: None, data_rate: 9600 }
0.000000 DEBUG Set dtr true, rts true
```
First part is `picotool reboot` and second is manual restart, note the timestamps

I'll try to submit a pull request for this today if I'm right and setting ticks up fixes it.

CBJamo, 2024-08-24:
Nice catch. When doing the initial support I loaded all my firmware using the uf2 method, and so the usb bootloader must've set up timer0 for me. Since the bootrom source is [available](https://github.com/raspberrypi/pico-bootrom-rp2350) it'd probably be a good idea to take a look through for anything else it was doing for me.

brandonros, 2024-09-08:
just calling this out for anybody https://hackaday.com/2024/09/04/the-worsening-raspberry-pi-rp2350-e9-erratum-situation/

ionspin, 2024-09-14:
I'm working on adding TRNG support, hopefully I have a draft this weekend.

sndnvaps, 2024-10-14:
It otc 14. Is any progress for the RP2350 Pico2...

CBJamo, 2024-10-14:
Embassy doesn't have board specific code, but embassy does support the rp2350, yes. Check out the examples here: https://github.com/embassy-rs/embassy/tree/main/examples/rp23

sndnvaps, 2024-10-15:
> Embassy doesn't have board specific code, but embassy does support the rp2350, yes. Check out the examples here: https://github.com/embassy-rs/embassy/tree/main/examples/rp23

i known it support, but not publish it to crates.io yet.
in 

```toml
embassy-rp = "0.2.0"
```
features not support `rp235xa'

CBJamo, 2024-10-15:
For now you'll have to use a git dependency. See here: https://embassy.dev/book/index.html#_how_do_i_switch_to_the_main_branch

sndnvaps, 2024-10-15:
> For now you'll have to use a git dependency. See here: https://embassy.dev/book/index.html#_how_do_i_switch_to_the_main_branch

thanks bro, i have do it now.

fatfingers23, 2024-11-20:
> I don't think this is correct:
>
> https://github.com/embassy-rs/embassy/blob/423e5d7655d038ead9ad7f5f448cabbd0b330660/embassy-rp/src/clocks.rs#L515 [...]

Hey, do you think you could expand on what you meant by `defmt-uart` if you get a chance? Looking at getting  some kind of logging working for the Pico 2. Is it a feature for a crate already in embassy or another project like [defmt-serial](https://github.com/gauteh/defmt-serial). I I've just been using the rp23 uart example to log over to the debug probe. Any projects or examples you could point me to would be greatly appreciated! Thank you!