embassy-rs/embassy #3247: Question - STM32G4 Dual bank vs embassy-boot

usbalbin, 2024-08-11:
Hi!

Sorry if this is the wrong place to ask.

Related #2309

From skimming through https://blog.drogue.io/firmware-updates-part-1/ there seems to be quite some overlap between what embassy-boot does(or did when the blog post was written) when swapping firmwares vs what the stm32g4 MCUs with dual bank flash can do with simply setting one bit. I assume there are probably more MCUs with similar features.

Then there is the whole RWW thing. Does that mean that the regular program will be less affected performance wise while a new firmware is simultaneously being written to flash? If so, that would then mean that a latency sensitive program will be at great risk when not using this feature? Also, I see embassy-boot now supports having the "DFU" partition in externa flash, I assume that would not work with the above. 

Then there is the whole 'mark the firmware as successfully booted' which I assume is not handled by the magic of dual bank flash.

Is this hardware feature used by embassy-boot when available? If not, then when using such an MCU, what would make most sense?

* Just use embassy-boot as is
* Build something simpler around the dual bank features, perhaps using embassy-boot's library where applicable

Sorry if this is bad/the wrong/too broad questions to ask with no general answers. I am just trying to get some sort of feel for things :)

badrbouslikhin, 2024-08-27:
Hi,

I can't help with the RWW-related stuff since I haven't done any benchmarks myself, but I'll do my best to address the other points.

> Also, I see embassy-boot now supports having the "DFU" partition in externa flash, I assume that would not work with the above.

What do you mean when you say "that would not work with the above"?
I believe the current design is flexible enough that partitions can live anywhere.

> Is this hardware feature used by embassy-boot when available? If not, then when using such an MCU, what would make most sense?

As far as I know, embassy-boot doesn't currently use this hardware feature. I wasn't part of the original embassy-boot design team, but I believe they went for an MCU-agnostic approach to keep things flexible.

I haven't looked into implementing the dual-bank native switch myself. That said, my experience with embassy-boot has been pretty great so far!
If you take a stab at the implementation, I'd be interested to check the PR :).

usbalbin, 2024-08-29:
> What do you mean when you say "that would not work with the above"?
I believe the current design is flexible enough that partitions can live anywhere.

Sorry. In my comparison between this bootloader vs the dual bank feature i meant that the dual bank feature by it self is not sufficient for using external flash for storing for example the non-active firmware. (Since the hardware only remaps the internal flash's banks)

> As far as I know, embassy-boot doesn't currently use this hardware feature. I wasn't involved in the original embassy-boot design, but but I'd guess they went with an MCU-agnostic approach to keep things flexible.

Makes sense :)

> If you take a stab at the implementation, I'd be interested to check the PR :).

Still not quite sure how to fit everything together. I will certainly look into it further down the line, thanks :)