embassy-rs/embassy #3250: TM1637 display protocol - desired for Embassy

marxin, 2024-08-12:
Just recently, I made a working TM1637 display using the GPIO ports with the Embassy framework. The code is mostly based on this: https://rust-classes.com/chapter_embedded_pi_input_tm1637.

Is the community interested in upstreaming the protocol into the Embassy?

lulf, 2024-08-15:
Rather than upstreaming that in embassy, it is generally preferable to implement drivers using traits from the various [embedded-hal](https://github.com/rust-embedded/embedded-hal) crates, which will allow the driver to work with not only Embassy but other HALs and other runtimes.

marxin, 2024-08-16:
Then, there's probably existing an implementation that depends on `embedded-hal`: https://github.com/igelbox/tm1637-rs

closed 2024-08-16 by marxin