embassy-rs/embassy #3256: STM32 Dual Core Init - Private Peripheral Bus considerations

liarokapisv, 2024-08-14:
The current dual-core initialization architecture assumes that system initialization is performed fully by the primary core.

Crucially this is not true for Peripherals addressed through the [Private Peripheral Bus](https://developer.arm.com/documentation/100166/0001/Programmers-Model/Processor-memory-model/Private-Peripheral-Bus), which includes the `NVIC`, `MPU` and `SysTick` through the [System Control Space](https://developer.arm.com/documentation/ddi0403/d/System-Level-Architecture/System-Address-Map/System-Control-Space--SCS-?lang=en).

Accesses to the [NVIC](https://github.com/rust-embedded/cortex-m/blob/b0d2e45a9a827f2c229340189cf125a35581f422/cortex-m/src/peripheral/mod.rs#L490) (and similarly for the other PPB peripherals), access a different instance depending on the core on which we are running.

As a result, this affects the [InterruptExt trait](https://github.com/embassy-rs/embassy/blob/7a26e117ccd5f8669548cc8c2424be4691c1c402/embassy-hal-internal/src/interrupt.rs#L153) and transively the [DMA init](https://github.com/embassy-rs/embassy/blob/7a26e117ccd5f8669548cc8c2424be4691c1c402/embassy-stm32/src/dma/dma_bdma.rs#L259) which ends up being called only on the `primary` core. This leaves the secondary core's `NVIC` with masked interrupts. The solution to this is most likely to duplicate the interrupt enablement on the `secondary_init` flow as well.

This interrupt masking is just one instance of the PPB issue and we will probably need to think things through about the rest of the `PPB` peripherals when it comes to dual core usage.

liarokapisv, 2024-08-14:
Going through the code it seems the biggest culprits are the enabled-at-initialization `dma` and `exti` modules.

leftger, 2026-10-09:
This looks like it was addressed by 2b7e76efe ("Fix dma nvic issues on dual core lines").

That commit stopped enabling DMA NVIC interrupts up-front on dual-core parts, and instead carries the interrupt per channel in `ChannelInfo` under the `_dual-core` feature, enabling it where it's actually needed. That structure is still present today (the `irq` field is emitted in `embassy-stm32/build.rs` and consumed under `src/dma/`).

Reopen or comment if the private-peripheral-bus behaviour you described is still reproducible.